tue-robotics/tue_robocup #501: Better 3d camera's

Rayman, 2017-11-21:
Can we see shiny and translucent objects better?

fibonatic, 2017-11-21:
I did a quick search and found this article: https://link.springer.com/chapter/10.1007/978-3-319-13560-1_33
I only skimmed through it, but it seems to say that they where able to improve seeing shiny and transparent objects.

LoyVanBeek, 2017-11-21:
Could be useful: https://roscon.ros.org/2017/presentations/ROSCon%202017%203D%20Vision%20Technology.pdf

MatthijsBurgh, 2017-11-21:
Not doing this, because this would need removal of all rgbd.

closed 2017-11-21 by MatthijsBurgh

LoyVanBeek, 2017-11-22:
Agreed, but this makes the rgbd package a major blocker for (potential) progress, which is a big risk too with the Kinects getting less and less available. Replacing RGBD and the kinect should be a goal next year.

MatthijsBurgh, 2017-12-05:
We have an Intel Realsense ZR300

LoyVanBeek, 2017-12-06:
Huh, how did that happen?

MatthijsBurgh, 2017-12-06:
@hvrooy buying one

alberth, 2017-12-06:
Nah, early Christmas present :)