tue-robotics/tue_robocup #504: Refactor ED to do segmentation in separate ROS service

LoyVanBeek, 2017-11-21:
Allows to more easily try out different segmentation methods.

reinzor, 2017-11-25:
How do you want to take the prior into account?

LoyVanBeek, 2017-11-25:
I don't know yet :smiley:. The original issue mentions this to be a high effort issue...

First guess is to implement most of https://github.com/tue-robotics/ed_sensor_integration/blob/master/src/kinect/segmenter.cpp as a ROS service. 
That would mean the objects used in those function call would need to be transmitted over ROS to the service, i.e. the world model objects (at least those in view) and some RGBD images.

MatthijsBurgh, 2017-12-02:
I agree with Rein. This is too difficult/too much work to separate. It might be useful to have the ability to segment and classify with a test setup.

closed 2017-12-05 by MatthijsBurgh