tue-robotics/tue_robocup #505: Find literature on translucent object segmentation

LoyVanBeek, 2017-11-21:
There was some RoboCup@Home team a couple of years ago that said they could do this. 

Find out how they do it and see if we can use the same.

MatthijsBurgh, 2017-12-02:
Homer TDP 2015:
Transparent objects (in our case drinking glasses) are detected by making use of one fault of the structured light sensor. The light emitted by the Xtion camera is scattered in transparent objects, providing no valid depth information. We segment areas with no depth data and compare them with drinking glass contour templates. Since the supporting table plane around the transparent objects has valid depth information, a size and location estimation of the transparent objects is obtained and used for grasping.

LoyVanBeek, 2017-12-03:
Good find! Can we use this approach in some way?

MatthijsBurgh, 2018-01-12:
No easy implementation in current pipeline possible. Performance of detection of bottles is improved by https://github.com/tue-robotics/ed_sensor_integration/pull/14.

closed 2018-01-12 by MatthijsBurgh