tue-robotics/tue_robocup #509: Check influence of margin around segmented object

LoyVanBeek, 2017-11-21:
As an easy fix for bad segmentation: does classification improve when the image is not cropped strictly around the object's edges as reported by ED, but take some margin around that.

This can be tested for feasibility manually first. E.g. take a large margin aroudn the object and automatically crop off e.g. 5%, 10%, 15& etc with something like image-magick (```find . -name '*.jpg' -execdir mogrify -crop 90%x+0+0 -gravity Center {} \;```)

alberth, 2017-11-22:
Random idea, what if you enlarge the image by copying the edge pixels repeatedly?
Would only work if there are no object pixels there, though

LoyVanBeek, 2017-11-22:
In that case, there is no information added or removed. 

Segmentation should take care of taking only the data/pixels relevant for the object itself and not the context. A margin adds context and the intuition is that that is bad because we want to correctly recognize objects regardless of context (ie. recognize them in the bookcase just as well as the training on only the dinner table, for example).

In training, there should be no context, ideally, because we do not want the classifier to learn an association between the object and it's background.

LoyVanBeek, 2017-11-26:
Results with the images from MEGA/data/robotics_testlabs/training_data_Josja/data_lab_large_margin, model is MEGA/data/robotics_testlabs/training_data_Josja/models_14/output_graph.pb

The best classifications seems to be in the 85-75% crop range. 

### 100%  crop, i.e. original images

[image]

### 95% crop
[image]

### 90% crop
[image]

### 85% crop
[image]

### 80% crop
[image]

### 75% crop
[image]

### 70% crop
[image]

### 65% crop
[image]

### 60% crop
[image]

### 55% crop
[image]

### 50% crop
[image]

### 45% crop
[image]

### 40% crop
[image]

MatthijsBurgh, 2017-12-05:
What is defined as 100%? Which ratio is similar to the images of ED?

LoyVanBeek, 2017-12-06:
100% is how @JosjaG  cropped the images manually. I'll upload some example pictures of what each crop looks like later. 

It seems however that how ED crops the images now is quite close to the optimum, but this is just my observation.

LoyVanBeek, 2017-12-06:
100%
[image]

90%
[image]

80%: More or less the optimum, so tightly cropped to the bounding box of the object itself. 
[image]

70%
[image]

60%
[image]

50%
[image]

40%:
[image]

LoyVanBeek, 2017-12-09:
Next up: check how ED currently does segmentation and adjust the margin it uses to the optimum we found.

closed 2017-12-19 by LoyVanBeek