tue-robotics/tue_robocup #510: Compare the accuracy of a classifier when trained on N classes. Does the accuracy diminish with increasing N classes and how much?

LoyVanBeek, 2017-11-21:


JosjaG, 2017-12-09:
Different sets trained on 3, 6, 9, 12 and 15 of the classes from /home/amigo/MEGA/data/robotics_testlabs/training_data_Josja/training:
3 classes:
[image]
6 classes:
[image]
9 classes:
[image]
12 classes:
[image]
15 classes:
[image]
Increasing the number of classes does lead to a decrease in the accuracy, as expected. This means that it might be a good idea to limit the number of classes if we don't have a lot of data.

MatthijsBurgh, 2018-01-13:
Conclusion: Reduce number of classes. So remove classes, which can't be detected. (Maybe also the classes that are always classified wrong/always have a too low probability.)

closed 2018-01-13 by MatthijsBurgh