tue-robotics/tue_robocup #516: Object_recognition NoneType

MatthijsBurgh, 2017-11-21:
[ERROR] [1511300831.634396]: Error processing request: 'NoneType' object has no attribute 'shape'
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/amigo/ros/kinetic/system/src/tensorflow_ros/scripts/object_recognition_node", line 85, in _recognize_srv_callback\n    size = self._bgr_image.shape[:2]  # For now, we assume the entire image is the ROI\n', "AttributeError: 'NoneType' object has no attribute 'shape'\n"]
[ERROR] [1511300831.635417530]: Service call failed: service [/amigo/object_recognition/recognize] responded with an error: error processing request: 'NoneType' object has no attribute 'shape'
[ERROR] [1511300831.635474074]: Service call failed

LoyVanBeek, 2017-11-22:
Which version is this error in? This code has a major refactor that was merged exactly when you filed this issue (https://github.com/tue-robotics/image_recognition/pull/30)

MatthijsBurgh, 2017-11-22:
Probably the old version. I am not sure though

LoyVanBeek, 2017-11-23:
That's the old version, which the scary multithreading :-). 
The problem is that ```self._bgr_image.shape``` is ```None```, I think. That member variable is completely gone now.

closed 2017-11-23 by LoyVanBeek