tue-robotics/tue_robocup #521: Tensorflow training parameters

MatthijsBurgh, 2017-12-05:


Rayman, 2017-12-05:
What is the issue?

MatthijsBurgh, 2017-12-05:
I know that we are now training with a batch size of 1000. Can we train with different batch sizes and what is the influence of it?

reinzor, 2017-12-05:
Please provide a proper description of the issue, only this title is not clear.

MatthijsBurgh, 2017-12-05:
Description added.

LoyVanBeek, 2017-12-06:
Yes we can train with that, influence on the performance is not known exactly.

Measuring that should be straightforward though.

Rayman, 2017-12-06:
I quote from [stackoverflow](https://stats.stackexchange.com/questions/164876/tradeoff-batch-size-vs-number-of-iterations-to-train-a-neural-network) and I also observed this in practice.
> It has been observed in practice that when using a larger batch there is a significant degradation in the quality of the model, as measured by its ability to generalize.

Conclustion use a small (1-10, or maybe 100) batch size.

MatthijsBurgh, 2017-12-06:
Lets train the network on like ~3 batch size. Run the tools from Loy and determine a batch size. Maybe one more iteration around the best result from the first set.

LoyVanBeek, 2017-12-19:
For some background: http://cs231n.github.io/neural-networks-3/#baby 

TL;DR: when validation accuracy starts going down while train accuracy goes up, then the network is likely to be overfitting (but this is not the only symptom). 

Also, when the accuracies reach a plateau, I'd say continuing to train is useless.

closed 2018-10-17 by MatthijsBurgh