tue-robotics/tue_robocup #523: Amigo is unable to grasp

MaxBaeten, 2017-12-05:
Even though it recognizes stuff it continously tries to grasp but then fails.

Hardware logs provides this error:
Dec 05 21:23:39 amigo1 sudo[989]: 24186.465 [ Warning][Logger] Gripper is OPEN - pos
Dec 05 21:23:41 amigo1 sudo[989]: 24188.091 [ Warning][Logger] Stopped driving because the reference-flow was interrupted.
Dec 05 21:23:46 amigo1 sudo[989]: [ERROR] [1512505426.198092459]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_pitch_joint_right elbow_roll_joint_right shoulder_pitch_joint_right shoulder_roll_joint_right shoulder_yaw_joint_right torso_joint wrist_pitch_joint_right wrist_yaw_joint_right ]
Dec 05 21:23:46 amigo1 sudo[989]: [ERROR] [1512505426.198139612]: Known controllers and their joints:
Dec 05 21:23:46 amigo1 sudo[989]: 
Dec 05 21:23:46 amigo1 sudo[989]: 24193.069 [ Info   ][Logger] TrajectoryActionlib: Received Message
Dec 05 21:23:46 amigo1 sudo[989]: 24193.076 [ Warning][Logger] Grippercontrol: received gripper Command: Close

Rayman, 2017-12-08:
How can we reproduce this. Which challenge?

MatthijsBurgh, 2017-12-08:
Today, the demo worked fine. Grasped stuff from dinner_table and cabinet

MatthijsBurgh, 2018-01-12:
Moving back to the backlog. I haven't seen this in a few weeks.

closed 2018-12-01 by MatthijsBurgh