tue-robotics/tue_robocup #524: AMIGO2 not responding

MatthijsBurgh, 2017-12-09:
I have experienced several times, that a terminal to AMIGO2 is not responding when not used for a few minutes.
I also noticed problems with the trigger topic. AMIGO doesn't respond to the trigger, when you ping or open a new ssh session. AMIGO does respond again.

https://github.com/tue-robotics/tue-env/pull/196 might show more insight in this, but this is not the cause, so it isn't solving the issue.

MatthijsBurgh, 2017-12-09:
@Rayman do you also have some knowledge to solve this?

Timple, 2017-12-09:
Can you reverse the settings of quickssh?

    type quickssh

MatthijsBurgh, 2017-12-09:
```
quickssh is aliased to `echo "ControlMaster auto
ControlPath /tmp/%r@%h:%p
ControlPersist yes" > ~/.ssh/config'
```

Timple, 2017-12-09:
Set ControlPersist No in ssh config, remove the /tmp/amigo@amigo2 file and try again

MatthijsBurgh, 2017-12-09:
What should be the behavior now?

Timple, 2017-12-10:
I don't know? Error gone? ;-)
This was something I could imagine that couples one ssh session to another (like the behaviour you describe)

MatthijsBurgh, 2017-12-10:
I think the error is not in the ssh. Because I also mentioned that AMIGO is not responding to the trigger topic. Which shows that is a more general network issue. I haven't paid attention to rviz, if it still shows current data, when there is no response. This could provide information about it being AMIGO2 or just AMIGO in general.

closed 2020-02-02 by LarsJanssenTUe