tue-robotics/tue_robocup #528: Unable to grasp anything

rokusottervanger, 2017-12-14:
Amigo inspects, segments, classifies, navigates to the object, moves its arm to the pre-grasp configuration, but then fails to grasp anything. 

In the log of amigo-start.service:
```
Dec 14 12:55:56 amigo1 sudo[980]: [ERROR] [1513252556.044586722]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_pitch_joint_right elbow_roll_joint_right shoulder_pitch_joint_right shoulder_roll_jo
Dec 14 12:55:56 amigo1 sudo[980]: [ERROR] [1513252556.044611868]: Known controllers and their joints:

```

jlunenburg, 2017-12-14:
I'll see if I can find anything next Tuesday. Tried it on Saturday and everything worked find (grasped and placed 2 out of 2 objects)

rokusottervanger, 2017-12-14:
After a tue-get update and tue-make everything worked again... I'm guessing someone ran a system update on amigo1. I'll close this for now.

closed 2017-12-14 by rokusottervanger

MatthijsBurgh, 2017-12-14:
I think this wasn't the first time this error showed up. So lets monitor this.