tue-robotics/tue_robocup #536: SPR: SSL not connected

jlunenburg, 2018-01-02:
If amigo1 is rebooted but the matrix is not, the ROS topic is not connected. As a result, the robot does not rotate in the correct direction and therefore does not understand any questions. Multiple solutions can be cooked up:
- upon startup of the challenge, check if the ssl topic is connected (robot.ssl._sub.get_num_connections() > 0)
- SSL node: if not connected to roscore, try to reconnect (somehow)

MatthijsBurgh, 2018-01-12:
@JosjaG will (have) build in a check at the challenge side. Somebody else, probably @reinzor, will have to build in a reconnection mechanism at the node side.

MatthijsBurgh, 2018-01-16:
@JosjaG has created a check in the challenge, which works good! 👍 But @LoyVanBeek created a more general check in #546. No fix in the ssl node yet.