tue-robotics/tue_robocup #563: Openface crashed when recognizing operator

KevinDang74, 2018-02-03:
It seems a face was detected correctly but that openface or lua is failing/not correctly configured. 

```
[INFO] [1517667378.463612]: 'Please look at me while I learn to recognize you.'
[INFO] [1517667381.304547]: Creating subscriber
[INFO] [1517667381.307774]: lazy subscribe to /amigo/top_kinect/rgb/image
[INFO] [1517667381.308106]: getting one image...
[INFO] [1517667381.308574]: self._camera_last_image: None
[INFO] [1517667381.472585]: len(self._camera_last_image): 3507018
[INFO] [1517667381.473056]: got 3507018 bytes of image data
[INFO] [1517667384.236400]: found 1 face(s) in the image
[INFO] [1517667384.237011]: found 1 valid face(s)
[INFO] [1517667384.237384]: annotating that face as operator
[ERROR] [1517667385.622866]: annotate failed: service [/amigo/face_recognition/annotate] responded with an error: error processing request: Could not get representation of face image: Could not get representation of face image: 

OpenFace: `openface_server.lua` subprocess has died.

+ Is the Torch command `th` on your PATH? Check with `which th`.

+ If `th` is on your PATH, try running `./util/profile-network.lua`
  to see if Torch can correctly load and run the network.

  + If this gives illegal instruction errors, see the section on
    this in our FAQ at http://cmusatyalab.github.io/openface/faq/

  + In Docker, use a Bash login shell or source
     /root/torch/install/bin/torch-activate for the Torch environment.

+ See this GitHub issue if you are running on a non-64-bit machine:
  https://github.com/cmusatyalab/openface/issues/42

+ Advanced Users: If you think this problem is caused by
running Lua as a subprocess, Vitalius Parubochyi has created
a version of this that uses https://github.com/imodpasteur/lutorpy.
This file is available at <openface>/openface/torch_neural_net.lutorpy.py
and our mailing list discussion on this can be found at:
https://groups.google.com/forum/#!topic/cmu-openface/Jj68LJBdN-Y

+ Please post further issues to our mailing list at
  https://groups.google.com/forum/#!forum/cmu-openface

Diagnostic information:

cmd: ['/usr/bin/env', 'th', '/usr/local/lib/python2.7/dist-packages/openface/openface_server.lua', '-model', '/home/amigo/openface/models/openface/nn4.small2.v1.t7', '-imgDim', '96']

============

stdout: 

```

KevinDang74, 2018-02-03:
Error suggests checking ```which th```:
```
amigo@amigo2 $ which th
/home/amigo/torch/install/bin/th
```

JosjaG, 2018-03-06:
This error is caused by corrupted/missing files. These files aren't updated since the folders already exist so the installing/updating procedure is skipped. This can be solved by removing the empty/corrupted folder: ``` sudo rm -rf torch/ openface/ ``` and then updating: ``` tue-get update ```.

closed 2018-03-06 by JosjaG