tue-robotics/tue_robocup #564: In simulator help-me-carry empty arm could not be resolved

fibonatic, 2018-02-03:
[ERROR] [1517670502.782079]: ArmDesignator 

Designator(
  all_arms = OrderedDict([('right', Arm(side='right')), ('left', Arm(side='left'))])
  available = <bound method UnoccupiedArmDesignator.available of Designator(resolve_type=<class 'robot_skills.arms.Arm'>, name=empty_arm_designator)>
  lockable = <bound method UnoccupiedArmDesignator.lockable of Designator(resolve_type=<class 'robot_skills.arms.Arm'>, name=empty_arm_designator)>
  name = empty_arm_designator
  preferred_arm = Arm(side='right')
  resolve_type = <class 'robot_skills.arms.Arm'>)

 could not resolve to an arm
[ERROR] [1517670502.782515]: Could not resolve arm

LoyVanBeek, 2018-02-03:
Any context? When did this happen? Did the robot already have 2 things in it's hands? Were some arms unavailable due to failing HW?

fibonatic, 2018-02-03:
This message occurs at the state GRAB_ITEM state when running the help-me-carry challenge. Both arms are empty.

The GRAB_ITEM state has an empty_arm_designator as argument, but bag_arm_designator is used, which is defined as:

empty_arm_designator = ds.UnoccupiedArmDesignator(robot.arms,robot.rightArm,name="empty_arm_designator")

bag_arm_designator = empty_arm_designator.lockable()
bag_arm_designator.lock()

So I am not sure if maybe the lock might be messing with things.

fibonatic, 2018-02-20:
This issue has been resolved.

closed 2018-02-20 by fibonatic