tue-robotics/tue_robocup #569: [EPIC^2] SERGIO

MatthijsBurgh, 2018-02-27:
# SERGIO
## Architecture
### Decisions
- [ ] How to apply launch structure of AMIGO --> One PC with all IO connections? Which one?
### ToDo's
- [ ] Fix all sergio install targets
- [ ] Install pc's correctly

## Hardware
### Decisions
- [ ] Where to place the Jetson and Matrix Creator
### ToDo's
- [ ] Reinstall Kinect
- [ ] Place Jetson
- [ ] Place Matrix creator
- [ ] Fix "Static electricity issue"

## Low-level software
### Decisions
- [ ] Orocos vs Simulink?
- [ ] How to RS485? First implementation and Final implementation.
- [ ] TUeES030 firmware version, switch now? Or when attaching arm(s)?

### ToDo's

## Base
### Decisions

### ToDo's
- [ ] Replace fancy covers by old robust covers
- [ ] Fix base controller; Check with torso and dummy masses of arms.

## Torso
### Decisions

### ToDo's
- [ ] Fix static pose
- [ ] Check URDF model(Especially height of joints around Base-Torso connection)
- [ ] Check calibration procedure and calculation of joints from measurements
- [ ] Torso control
- [ ] Torso planning

## Arms
### Decisions

### ToDo's

## Grippers
### Decisions

### ToDo's

Timple, 2018-02-27:
Orocos vs Simulink **vs ROS**

Lets not leave out the options which does not require separate maintainers or knowledge.

Rayman, 2018-02-27:
http://wiki.ros.org/ros_control

LoyVanBeek, 2018-03-01:
Some useful links for ROS_control, SOEM, EtherCAT:

- https://openethercatsociety.github.io/doc/soem/index.html
- [minas_hardware_interface](https://github.com/tork-a/minas/blob/master/minas_control/src/minas_hardware_interface.cpp) and [ethercat_manager](https://github.com/tork-a/minas/blob/master/ethercat_manager/src/ethercat_manager.cpp) as an example of a different robot using EtherCAT
- [Another ethercat_manager](https://github.com/ros-industrial/robotiq/blob/jade-devel/robotiq_ethercat/src/ethercat_manager.cpp) by Robotiq
- [Shadow Robot EtherCAT interface](https://github.com/shadow-robot/sr-ros-interface-ethercat)
- [ROS EtherCAT also by Shadow Robotics](https://github.com/shadow-robot/ros_ethercat)

alberth, 2018-03-02:
While leaving last Tuesday, Max talked about considerations of organizing EtherCaT slave code. That got me thinking, I recently wrote a library for E-Box data acquisition on top of Soem that detects the slave setup (for more slave types than just E-Box), and provides elementary IO point access to whatever slave configuration you have. It's all in C89, as the code is the bottom of a (generated) MEX function, but that can be changed if necessary. I didn't implement the IO recovery procedure, since in the use-case, it's simply 'game over' in case of IO failure.

Timple, 2018-03-03:
At Nobleo Projects we heb been working on an ethercat interface as well. The once mentioned above did not fulfill our needs (yet). Although I though I only found 4 out of 5.

We where using a working ros_control implementation for our AGV. However the ros_control was not very dynamically scalable. So we are in the process of migrating to a node which dynamically detects the slaves and exposes interfaces for each slave. Actually quite similar to the orocos node.

The source code can be found here:
https://bitbucket.org/nobleo/ethercat_interface/src/single-node/

If desired and usefull I am willing to maintain an github repo for this as well.