tue-robotics/tue_robocup #572: CanTransform timeout error when transforming region of interest to map

JosjaG, 2018-03-14:
When attempting to transform a kinect image to the map, the transform receives a timeout error and everything dies: 
``` 
[INFO] [1520965397.318223]: project_rois response: points: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1520965397
        nsecs:  97251654
      frame_id: /amigo/top_kinect/openni_rgb_optical_frame
    point: 
      x: -0.409936810987
      y: -0.0194662409335
      z: 2.31227111816
Transforming roi to map
[ERROR] [1520965398.328980]: head.project_roi failed: . canTransform returned after 1.01006 timeout was 1.
```

From code:
``` 
rospy.loginfo("best_detection = {}".format(best_detection))
if best_detection:
   roi = best_detection.roi
   try:
      operator_pos_kdl = self._robot.perception.project_roi(roi=roi, frame_id="map")
   except Exception as e:
       rospy.logerr("head.project_roi failed: %s", e)
       return 'Failed'
```
The error always occurs in real life testing, and only sometimes in simulation.

LoyVanBeek, 2018-03-27:
Complete example for ```amigo-console```

```python
raw_detections = amigo.perception.detect_faces()
best_detection = amigo.perception.get_best_face_recognition(raw_detections, "operator")

rospy.loginfo("best_detection = {}".format(best_detection))
if best_detection:
   roi = best_detection.roi
   try:
      operator_pos_kdl = amigo.perception.project_roi(roi=roi, frame_id="map")
   except Exception as e:
       rospy.logerr("head.project_roi failed: %s", e)
       return 'Failed'
```

LoyVanBeek, 2018-03-27:
The call that fails is ```robot.perception.project_roi```` The simplest way to call that is with (in ```amigo-console```)
```
from sensor_msgs.msg import RegionOfInterest

amigo.perception.project_roi(RegionOfInterest(593, 619, 149, 128, False), "/map")
```

LoyVanBeek, 2018-03-27:
Call stack: 

1. https://github.com/tue-robotics/tue_robocup/blob/master/robot_skills/src/robot_skills/util/kdl_conversions.py#L71
2. https://github.com/ros/geometry/blob/indigo-devel/tf/src/tf/listener.py#L71 
3. https://github.com/ros/geometry/blob/indigo-devel/tf/src/tf/listener.py#L74 
4. https://github.com/ros/geometry2/blob/melodic-devel/tf2_ros/src/tf2_ros/buffer.py#L86
5. https://github.com/ros/geometry2/blob/melodic-devel/tf2_ros/src/tf2_ros/buffer.py#L87
6. ?
7. https://github.com/ros/geometry2/blob/5722773de7d688bcaf8dfe68a0f2a55ff49dc9d3/tf2_ros/src/buffer.cpp#L112 is where the message ". canTransform returned after"... comes from.

MatthijsBurgh, 2018-04-09:
Solved by switching to TF for tf listener instead of tf_server. right?

LoyVanBeek, 2018-04-10:
No, the issue is with normal tf_listener, not tf_server.

MatthijsBurgh, 2018-12-01:
@LoyVanBeek did we fix it in the end? I haven't seen this bug in ages.

MatthijsBurgh, 2019-03-17:
@LoyVanBeek fixed?

LoyVanBeek, 2019-03-19:
I haven't seen it in ages either.

closed 2019-03-19 by LoyVanBeek