tue-robotics/tue_robocup #577: Simulator SERGIO

MatthijsBurgh, 2018-04-09:
# Deliverable
- [ ] Working Simulator with working first stage challenges

# Steps:
- [ ] SSL
- [ ] Grasping
- [ ] Navigation
- [ ] Head

LoyVanBeek, 2018-04-10:
In simulation, rosnode ```/sergio/head_ref``` publishes on ```/sergio/neck/references```. 
However, the rosnode ```sergio/fast_simulator``` only has an action_server ```/sergio/body/joint_trajectory_action```, which includes the ```neck_pan_joint, neck_tilt_joint```
This is one mismatch. 
There is another: sergio_description/custom/joint_groups.yaml defines
```yaml
    - name: head
      reference_topic: "/sergio/head/references"
      measurement_topic: "/sergio/head/measurements"
      joint_names:
          - neck_pan_joint
          - neck_tilt_joint
```
so the neck joints are expected on /sergio/head/references while ```/sergio/head_ref``` publishes on ```/sergio/neck/references```.

LoyVanBeek, 2018-04-10:
The fast_simulator defines Robo-suubclasses for both Amigo and Sergio. 
For Amigo, it subscribes to 
```cpp
sub_head = nh.subscribe("/amigo/neck/references", 10, &Amigo::callbackJointReference, this);
``` while for Sergio, there is no such subscriber.

LoyVanBeek, 2018-04-17:
On the real robot, there is nothing that listens to ```/sergio/body/joint_trajectory_action/...```. 

For the neck, the joints will not be in  ```/sergio/body/joint_trajectory_action/...``` but in a separate  ```/sergio/neck/joint_trajectory_action/...```. This does not yet exist, neither in simulation nor real-robot. 

@Rayman will make a node that listens to ```/sergio/neck/joint_trajectory_action``` and control these joints, in both real-life and simulation. In the real robot, this will be a ros_control meta-controller. 

As to  ```/sergio/body/joint_trajectory_action```, that requires a sergio_moveit_config.

LoyVanBeek, 2018-04-24:
All in  all, these PRs need to be merged to make this bit work:
- https://github.com/tue-robotics/head_ref/pull/9
- https://github.com/tue-robotics/fast_simulator/pull/10
- https://github.com/tue-robotics/sergio_description/pull/6

rokusottervanger, 2018-04-25:
We will get rid of the neck control in Orocos for AMIGO, and use the USB dynamixel controllers like on SERGIO. This is implemented on https://github.com/tue-robotics/amigo_bringup/pull/95

LoyVanBeek, 2018-04-25:
Status:
- With the branches mentioned above, head_ref works in sim for amigo & sergio and for real on Sergio but not yet tested on amigo.

On Sergio, the head seems to jitter in  weird way: when the head is still, sometimes a different measurement comes through. That different one is also stable but only sent intermittently.