tue-robotics/tue_robocup #579: Chrony does not sync Amigo1 and Amigo2 properly

rokusottervanger, 2018-04-11:
After a restart of the chrony service on amigo2, they sync again. But on startup they don't.

Timple, 2018-04-12:
The problem with these intermittent issues is that they are hard to debug.

@rokusottervanger 
- Did this occur once or more often?
- How did you notice the problem?

rokusottervanger, 2018-04-12:
- As I understand, this happened consistently in the last few days. 
 - A graduation student was trying to get his own executive working, but was testing the demo mode first. He could not get Kinect images until after a restart of the chrony service on amigo2 (clocks were out of sync by a few minutes).
 - This happened last friday morning and last monday morning at least.

Timple, 2018-04-12:
Good do know it reproduces kind of. Might we get the output of `chronyc tracking` of both amigos next time this occurs? Before restarting anything obviously.

LoyVanBeek, 2018-04-25:
We had a similar problem on Sergio today. Solution was to allow sergio2's IP address in the chrony.conf of sergio1.

MatthijsBurgh, 2019-01-08:
I have not been experiencing this issue. Rebooting AMIGO2 always solves the problem for me.

closed 2019-01-08 by MatthijsBurgh