tue-robotics/tue_robocup #581: no diagnostic msg received for the right_arm pops up all the time since on real robot no arms (and possibly spindle/torso)

SamAlexandrov, 2018-04-17:
[WARN] [1523979999.902044]: no diagnostic msg received for the right_arm
[WARN] [1523979999.902138]: no diagnostic msg received for the right_arm
[WARN] [1523979999.902448]: no diagnostic msg received for the left_arm
[WARN] [1523979999.902496]: no diagnostic msg received for the left_arm
[WARN] [1523979999.902764]: no diagnostic msg received for the spindle
[WARN] [1523979999.902765]: no diagnostic msg received for the spindle
[WARN] [1523980000.001558]: no diagnostic msg received for the head
[WARN] [1523980000.001561]: no diagnostic msg received for the head
[WARN] [1523980000.001923]: no diagnostic msg received for the right_arm
[WARN] [1523980000.001950]: no diagnostic msg received for the right_arm
[WARN] [1523980000.002183]: no diagnostic msg received for the left_arm
[WARN] [1523980000.002212]: no diagnostic msg received for the left_arm
[WARN] [1523980000.002448]: no diagnostic msg received for the spindle
[WARN] [1523980000.002486]: no diagnostic msg received for the spindle

LoyVanBeek, 2018-04-17:
This is because:

1. https://github.com/tue-robotics/tue_robocup/blob/master/robot_skills/src/robot_skills/robot.py#L76 calls 
2. https://github.com/tue-robotics/tue_robocup/blob/master/robot_skills/src/robot_skills/robot.py#L196 often. That calls 
3. https://github.com/tue-robotics/tue_robocup/blob/master/robot_skills/src/robot_skills/robot_part.py#L167 which generates that message. 

The problem is that in simulation, Sergio does have e.g. arms (which it can't yet use) but because the hardware in real life misses these components, there are no diagnostic messages for those and the crresponding body parts complain that there no message for them.

LoyVanBeek, 2018-04-17:
Most correct solution would be to remove these parts from Sergio but that requires some refactoring. Some challenges then also won't work because those should check presence of the body parts they need. 

Practically, we could just ignore some body parts in Sergio's diagnostics for now.

LoyVanBeek, 2018-04-19:
There's little point to that. The code was based on the assumption that missing diagnostic info for a part was a rare occurrence, not the default.

closed 2018-06-02 by Rayman