tue-robotics/tue_robocup #598: Detect table instead of the biggest object

jlunenburg, 2018-05-08:


MatthijsBurgh, 2018-05-27:
Maybe we should start by fixing the score (https://github.com/tue-robotics/tue_robocup/blob/master/challenge_restaurant/src/challenge_restaurant/store_waypoint.py#L127). Currently the side with the highest score is chosen. The score is the sum minimal distance and area sum of all detected objects on that side. Which is not logical, because if the same table is detected on both sides, than the side with the table furthest away is chosen. Plus taking into account the area of all detected object is also not completely logical IMO, but the minimal distance is really causing bad decisions.

Example:
Robot was standing next to dinnertable in RGO2018 arena. So table is at right. This example also shows that the area of the table is not calculated, so the shape of the entity is not seen as a rightprism (https://github.com/tue-robotics/tue_robocup/blob/master/challenge_restaurant/src/challenge_restaurant/store_waypoint.py#L109)
```
[INFO] [1527397961.926722]: looking at side right
[INFO] [1527397964.982944]: Updating kinect for side right
[WARN] [1527397964.983392]: No area_description provided for 'update_kinect'. This is probably a bad idea.
Succesfully stored '/home/amigo/ed/kinect/2018-05-27/2018-05-27-01-12-44_rgb.png'
[INFO] [1527397965.948307]: Found 3 entities for side right
[INFO] [1527397967.050519]: looking at side left
[INFO] [1527397972.511282]: Updating kinect for side left
[WARN] [1527397972.511678]: No area_description provided for 'update_kinect'. This is probably a bad idea.
Succesfully stored '/home/amigo/ed/kinect/2018-05-27/2018-05-27-01-12-52_rgb.png'
[INFO] [1527397973.448475]: Found 2 entities for side left
[INFO] [1527397974.552280]: Side right scoring (0.000000): {'min_distance': 0.0, 'area_sum': 0}
[INFO] [1527397974.553091]: Side left scoring (0.258976): {'min_distance': 0.21462801491826453, 'area_sum': 0.044347676529785912}
```

MatthijsBurgh, 2018-06-25:
New table detection implemented by @reinzor on RWC2018. This issue can be closed if `robocup` branch is merged in master.

MatthijsBurgh, 2018-10-31:
#639 is merged, so closing.

closed 2018-10-31 by MatthijsBurgh