tue-robotics/tue_robocup #604: Update RoboCup git scripts

jlunenburg, 2018-05-12:
* Remove tue-set-github-origin functions. 
* Add tool to add server as remote, checkout new branches that track this new remote.

LoyVanBeek, 2018-05-12:
The way we use git at a tournament is utterly wrong :-). 
Instead over overwriting the address of the ```github``` remote, we should simply add another remote, called ```roboticssrv``` that points to ``roboticssrv.local```

At a tournament, we should make a branch for that tournament only and use that branch during that tournament. The default remote of that branch should be ```roboticssrv```. 

After a tournament, we push that branch from  ```roboticssrv.local``` to ```github.com```.

jlunenburg, 2018-05-15:
@LoyVanBeek maybe it's better to  focus on the challenges first (you need robots to test those) and to fix this as the robots are on their way...

jlunenburg, 2018-06-05:
@LoyVanBeek did you already get to this? @MatthijsBurgh already offered to help

MatthijsBurgh, 2018-06-08:
see tue-robotics/tue-env#267

closed 2018-06-16 by MatthijsBurgh