tue-robotics/tue_robocup #609: Slack & Telegram API key storage

LoyVanBeek, 2018-05-12:
How can we privately store the Telegram and Slack API keys and load them as parameters for the telegram_bridge and slack_bridge?

Those should be loaded only when running on the real robot, otherwise load some demo API key.

LoyVanBeek, 2018-05-12:
@reinzor any ideas how to do this well?

alberth, 2018-05-14:
Load it from the command line (what you have now), with a fallback to load it from a file that only exists at the robot, with a fallback to use a hard-coded demo key?

reinzor, 2018-05-14:
https://github.com/tue-robotics/slack_ros/blob/master/scripts/slack_ros_bridge#L18

Obtained via a ROS param. This param can be specified with use of a parameter file or via the command line. With use of `<group if="$(optenv ROBOT_REAL">` we can load different parameters on startup. See amigo start.launch. We can point to a file in the /MEGA folder, this is a choice.

LoyVanBeek, 2018-05-14:
Yeah, that was my idea as well. 
I don't have the Slack API keys though, can you add them to ~/MEGA/credentials/slack/token_....yaml, @reinzor ? For now  I added a single test keys for test/amigo/sergio.

LoyVanBeek, 2018-05-14:
https://github.com/tue-robotics/robot_launch_files/pull/28

closed 2018-05-21 by LoyVanBeek