tue-robotics/tue_robocup #620: RWC 2018 aftermath

MatthijsBurgh, 2018-06-25:
- [ ] [action_server](https://github.com/tue-robotics/action_server) No actions taken yet
- [ ] [amigo_bringup](https://github.com/tue-robotics/amigo_bringup) No actions taken yet
- [ ] [amigo_description](https://github.com/tue-robotics/amigo_description) No actions taken yet
- [ ] [amigo_hardware](https://github.com/tue-robotics/amigo_hardware) No actions taken yet
- [ ] [amigo_moveit_config](https://github.com/tue-robotics/amigo_moveit_config) No actions taken yet
- [ ] [amigo_msgs](https://github.com/tue-robotics/amigo_msgs) No actions taken yet
- [ ] [amigo_ref_interpolator](https://github.com/tue-robotics/amigo_ref_interpolator) No actions taken yet
- [ ] [ar_marker_detector](https://github.com/tue-robotics/ar_marker_detector) No actions taken yet
- [ ] [audio_common](https://github.com/tue-robotics/audio_common) No actions taken yet
- [ ] [battery_manager](https://github.com/tue-robotics/battery_manager) No actions taken yet
- [ ] [cb_base_navigation](https://github.com/tue-robotics/cb_base_navigation) No actions taken yet
- [ ] [cb_planner_msgs_srvs](https://github.com/tue-robotics/cb_planner_msgs_srvs) No actions taken yet
- [ ] [code_profiler](https://github.com/tue-robotics/code_profiler) No actions taken yet
- [ ] [control_msgs](https://github.com/tue-robotics/control_msgs) No actions taken yet
- [ ] [conversation_engine](https://github.com/tue-robotics/conversation_engine) No actions taken yet
- [ ] [dashboard](https://github.com/tue-robotics/dashboard) No actions taken yet
- [ ] [diagnostics](https://github.com/tue-robotics/diagnostics) No actions taken yet
- [ ] [dialogflow_ros](https://github.com/tue-robotics/dialogflow_ros) No actions taken yet
- [ ] [dragonfly_speech_recognition](https://github.com/tue-robotics/dragonfly_speech_recognition) No actions taken yet
- [ ] [dynamixel_jointstate_conversion](https://github.com/tue-robotics/dynamixel_jointstate_conversion) No actions taken yet
- [ ] [dynamixel_motor](https://github.com/tue-robotics/dynamixel_motor) No actions taken yet
- [ ] [dynamixel_ros_control](https://github.com/tue-robotics/dynamixel_ros_control) No actions taken yet
- [ ] [ed](https://github.com/tue-robotics/ed) No actions taken yet
- [ ] [ed_gui_server](https://github.com/tue-robotics/ed_gui_server) No actions taken yet
- [ ] [ed_localization](https://github.com/tue-robotics/ed_localization) No actions taken yet
- [ ] [ed_moveit](https://github.com/tue-robotics/ed_moveit) No actions taken yet
- [ ] [ed_msgs](https://github.com/tue-robotics/ed_msgs) No actions taken yet
- [ ] [ed_navigation](https://github.com/tue-robotics/ed_navigation) No actions taken yet
- [ ] [ed_object_models](https://github.com/tue-robotics/ed_object_models) No actions taken yet
- [ ] [ed_perception](https://github.com/tue-robotics/ed_perception) No actions taken yet
- [ ] [ed_perception_models](https://github.com/tue-robotics/ed_perception_models) No actions taken yet
- [ ] [ed_robocup](https://github.com/tue-robotics/ed_robocup) No actions taken yet
- [ ] [ed_rviz_plugins](https://github.com/tue-robotics/ed_rviz_plugins) No actions taken yet
- [ ] [ed_sensor_integration](https://github.com/tue-robotics/ed_sensor_integration) No actions taken yet
- [ ] [ed_tutorials](https://github.com/tue-robotics/ed_tutorials) No actions taken yet
- [ ] [ed_wire](https://github.com/tue-robotics/ed_wire) No actions taken yet
- [ ] [emc-env](https://github.com/tue-robotics/emc-env) No actions taken yet
- [ ] [emc_simulator](https://github.com/tue-robotics/emc_simulator) No actions taken yet
- [ ] [emc_system](https://github.com/tue-robotics/emc_system) No actions taken yet
- [ ] [emergency_speakup](https://github.com/tue-robotics/emergency_speakup) No actions taken yet
- [ ] [ethercat_interface](https://github.com/tue-robotics/ethercat_interface) No actions taken yet
- [ ] [executive_smach](https://github.com/tue-robotics/executive_smach) No actions taken yet
- [ ] [fast_simulator](https://github.com/tue-robotics/fast_simulator) No actions taken yet
- [ ] [fast_simulator_data](https://github.com/tue-robotics/fast_simulator_data) No actions taken yet
- [ ] [geolib2](https://github.com/tue-robotics/geolib2) No actions taken yet
- [ ] [gitim](https://github.com/tue-robotics/gitim) No actions taken yet
- [ ] [gpsr_command_generator](https://github.com/tue-robotics/gpsr_command_generator) No actions taken yet
- [ ] [GPSRCmdGen](https://github.com/tue-robotics/GPSRCmdGen) No actions taken yet
- [ ] [grammar_parser](https://github.com/tue-robotics/grammar_parser) No actions taken yet
- [ ] [gripper_detector](https://github.com/tue-robotics/gripper_detector) No actions taken yet
- [ ] [head_ref](https://github.com/tue-robotics/head_ref) No actions taken yet
- [ ] [head_ref_ed_client](https://github.com/tue-robotics/head_ref_ed_client) No actions taken yet
- [ ] [hmi](https://github.com/tue-robotics/hmi) No actions taken yet
- [ ] [image_recognition](https://github.com/tue-robotics/image_recognition) No actions taken yet
- [ ] [image_recognition-release](https://github.com/tue-robotics/image_recognition-release) No actions taken yet
- [ ] [image_recognition_wire](https://github.com/tue-robotics/image_recognition_wire) No actions taken yet
- [ ] [kinect_driver](https://github.com/tue-robotics/kinect_driver) No actions taken yet
- [ ] [matrix_creator_ros](https://github.com/tue-robotics/matrix_creator_ros) No actions taken yet
- [ ] [multirobot_communication](https://github.com/tue-robotics/multirobot_communication) No actions taken yet
- [ ] [navigation](https://github.com/tue-robotics/navigation) No actions taken yet
- [ ] [nl_robot_console](https://github.com/tue-robotics/nl_robot_console) No actions taken yet
- [ ] [node_alive](https://github.com/tue-robotics/node_alive) No actions taken yet
- [ ] [openpose](https://github.com/tue-robotics/openpose) No actions taken yet
- [ ] [people_detector](https://github.com/tue-robotics/people_detector) No actions taken yet
- [ ] [picaso_4d_systems](https://github.com/tue-robotics/picaso_4d_systems) No actions taken yet
- [ ] [pico_base_controller](https://github.com/tue-robotics/pico_base_controller) No actions taken yet
- [ ] [pico_bringup](https://github.com/tue-robotics/pico_bringup) No actions taken yet
- [ ] [pico_talk](https://github.com/tue-robotics/pico_talk) No actions taken yet
- [ ] [qr_detector_zbar](https://github.com/tue-robotics/qr_detector_zbar) No actions taken yet
- [ ] [r5cop_demo](https://github.com/tue-robotics/r5cop_demo) No actions taken yet
[... 27 lines cut ...]
- [ ] [tensorflow_playground](https://github.com/tue-robotics/tensorflow_playground) No actions taken yet
- [ ] [text_to_speech](https://github.com/tue-robotics/text_to_speech) No actions taken yet
- [ ] [tf_checker](https://github.com/tue-robotics/tf_checker) No actions taken yet
- [ ] [tf_server](https://github.com/tue-robotics/tf_server) No actions taken yet
- [ ] [tue-env](https://github.com/tue-robotics/tue-env) No actions taken yet
- [ ] [tue-robotics.github.io](https://github.com/tue-robotics/tue-robotics.github.io) No actions taken yet
- [ ] [tue_config](https://github.com/tue-robotics/tue_config) No actions taken yet
- [ ] [tue_control](https://github.com/tue-robotics/tue_control) No actions taken yet
- [ ] [tue_control_rtt](https://github.com/tue-robotics/tue_control_rtt) No actions taken yet
- [ ] [tue_control_rtt_msgs](https://github.com/tue-robotics/tue_control_rtt_msgs) No actions taken yet
- [ ] [tue_control_tools](https://github.com/tue-robotics/tue_control_tools) No actions taken yet
- [ ] [tue_docs](https://github.com/tue-robotics/tue_docs) No actions taken yet
- [ ] [tue_ethercat_drivers](https://github.com/tue-robotics/tue_ethercat_drivers) No actions taken yet
- [ ] [tue_filesystem](https://github.com/tue-robotics/tue_filesystem) No actions taken yet
- [ ] [tue_hsr](https://github.com/tue-robotics/tue_hsr) No actions taken yet
- [ ] [tue_manipulation](https://github.com/tue-robotics/tue_manipulation) No actions taken yet
- [ ] [tue_manipulation_msgs](https://github.com/tue-robotics/tue_manipulation_msgs) No actions taken yet
- [ ] [tue_mobile_ui](https://github.com/tue-robotics/tue_mobile_ui) No actions taken yet
- [ ] [tue_msgs](https://github.com/tue-robotics/tue_msgs) No actions taken yet
- [ ] [tue_robocup](https://github.com/tue-robotics/tue_robocup) No actions taken yet
- [ ] [tue_serialization](https://github.com/tue-robotics/tue_serialization) No actions taken yet
- [ ] [tue_simulink_ethercat](https://github.com/tue-robotics/tue_simulink_ethercat) No actions taken yet
- [ ] [tue_teleop_keyboard](https://github.com/tue-robotics/tue_teleop_keyboard) No actions taken yet
- [ ] [virtual_cam](https://github.com/tue-robotics/virtual_cam) No actions taken yet
- [ ] [wire](https://github.com/tue-robotics/wire) No actions taken yet

MatthijsBurgh, 2018-06-25:
Lets move the following repos to the graveyard:
- [ ] [gpsr_command_generator](https://github.com/tue-robotics/gpsr_command_generator)
- [ ] [picaso_4d_systems](https://github.com/tue-robotics/picaso_4d_systems)
- [ ] [rwc2016_data](https://github.com/tue-robotics/rwc2016_data)
- [ ] [tf_checker](https://github.com/tue-robotics/tf_checker)
- [ ] [tue_control_rtt](https://github.com/tue-robotics/tue_control_rtt)
- [ ] [tue_control_rtt_msgs](https://github.com/tue-robotics/tue_control_rtt_msgs)
- [ ] [tue_control_tools](https://github.com/tue-robotics/tue_control_tools)

LoyVanBeek, 2018-07-26:
Allrighty, branches have been split up. Each challenge has a PR: rwc2018/<challenge_name>2 (the 2 because I forgot to include the relevant knowledge file originally"

@tue-robotics all: please create a PR from the branch you worked on, so others can review and approve of your work. 

The script I used to split the branches is at https://gist.github.com/LoyVanBeek/06cc66d8d3101ce539bf5264e2f2674b

closed 2019-05-08 by LarsJanssenTUe