tue-robotics/tue_robocup #649: Planning/Overview Robot Skills

JosjaG, 2018-10-02:
Responsible: @LoyVanBeek , @JosjaG 
- [ ] Implement Toyota MoveIt version for joint goals (instead of SafePoseChanger)
      - Note that this is currently done in the bridge nodes, moving/refactoring these bridge nodes should also be done.
- [ ] Refactor arms.py to make generic (no more left/right)
- [ ] Refactor moving the arms to be able to handle both joint goals and end effector poses (so on the high level only one way to move the arm and let the robot decide how)

jlunenburg, 2018-10-17:
`Arm.send_joint_goal` and `Arm.send_joint_trajectory` take a string as parameter. The corresponding joint position (or positions) are obtained from the (robot-specific) parameters by loading them from the parameter server at initialization of the Arm. In turn, the aforementioned methods call the 'private' `Arm._send_joint_trajectory` method. This 'private' method should *never* be used in the robot smach states because that results in robot-specific code.

alberth, 2018-10-22:
Sounds good. Do we assume all arms can do all string operations? If not, the arm designator selection code should be extended to select an arm with particular capabilities required by the challenge.

alberth, 2018-10-23:
Issues #679 and #680 address the problem of using the private method trajectory method

LoyVanBeek, 2018-11-13:
arms.py currently has a ```side``` parameter. If we change that to just an ```index```, I don't think we can go much further in making the arms itself separate from their side. 
We'll always need something to refer to specific action_clients or MoveGroups somehow, and I'd like to keep that a parameter given on construction.

There are no methods of Arm that take the ```side``` as a parameter; once constructed, it just knows that.

jlunenburg, 2018-11-27:
IMHO, not much needs to change in arms.py. We still need the namespacing of all the ROS stuff (although renaming `self.side` might give it a more generic impression. 

The real issues are in the `Robot` class. This adds (hardcoded) a `leftArm` and `rightArm` and that is definitely not applicable for HERO. A dict, however, would probably do just fine although the challenges need ways to return one of the arms. 

In the derived objects (`Amigo`, `Sergio`, `Hero`) it's still convenient to have properties for quick access to the arms when using the `amigo-console`. We have to consider, however, how we can prevent people from misusing these in the robot smach states and challenges

LoyVanBeek, 2018-11-27:
What we can do for that is assign ````leftArm``` and ```rightArm``` in the console only, not in the constructor. 
E.g. 
```python
class Banana(object):
    def __init__(self):
        pass

>>> b = Banana()    
>>> b.leftArm
AttributeError: 'Banana' object has no attribute 'leftArm'
>>> b.leftArm = Arm(...)
>>> b.leftArm
Arm(index='...')
```

Just assign it, but not in the constructor. 

This will be confusing though. With some magic (a property), you can add a warning to when you get tit directly (might be useful now too!) but that will be more magic and confusion.

MatthijsBurgh, 2018-12-22:
Is suggest a deadline of end of January. Is this possible?

LoyVanBeek, 2019-01-03:
https://github.com/tue-robotics/tue_robocup/pull/671 is a major step by @alberth For HERO we did make a step by merging some of it's branches. I ned to discuss with @JosjaG to estimate the work left.

MatthijsBurgh, 2019-01-08:
@alberth is fixing his stuff. The rest of this issue is on hold.

alberth, 2019-02-08:
Seems mostly done, except the last robot base class cleanup sneaked the arms back in, will fix that.

MatthijsBurgh, 2019-03-17:
Has the last issues been resolved?

alberth, 2019-03-19:
Fixing the problem is not complicated, except it makes `robot.arms` unavailable, so pretty much all challenges will be broken. As a result, the patch is expanding to address that part too.

I am not sure how long it will take to fix, probably it's better to use things as they are now, and continue, rather than waiting until it's all done.

closed 2021-01-27 by PetervDooren