tue-robotics/tue_robocup #655: HERO video material

MatthijsBurgh, 2018-10-06:
- [ ] Please list all video material you have available.
- [ ] Please list video material which we have of AMIGO, but not from HERO, and which could be filmed next Tuesday.

JosjaG, 2018-10-06:
In the lab:
- Navigation (combination of rviz and regular video)
- Picking up and placing objects
- Inspecting (rviz)
- ED web GUI (interactive map building) (combination of rviz and screen video)
- Telegram + hand over

In Arnhem (we have to check if we can use this):
- Navigation
- Picking up and handing over
- Picking up and placing
- Finding person in room

JosjaG, 2018-10-06:
What the AMIGO qualification video has that we don't:
- Video of rviz and simultaneously 
- Long videos (but that is probably a good thing :p)
- Imitation learning
- Separate object recognition (you do see it with the inspect, but that's not very clear)
- Robot-console (but we do show Telegram so I don't know if we need that)
- Small object detection
- Person/Face recognition
- Pose detection

JosjaG, 2018-10-06:
We could film:
- More live with simultaneous rviz video's
- Separate object and face recognition
- Small object detection

MatthijsBurgh, 2018-10-06:
Good busy!

JosjaG, 2018-10-09:
We made all the video's today, and I'll try to edit everything on Friday. I hope to have a draft finished by the end of the weekend!

closed 2018-10-29 by MatthijsBurgh