tue-robotics/tue_robocup #660: Speech Recognition

ar13pit, 2018-10-06:
- [ ] Complete installation of Kaldi-ASR
- [ ] Build a basic isolated word recognition system
- [ ] Update documentation
- [ ] Integrate with ```tue-env```
- [ ] Make ROS wrapper
- [ ] Interface the system with HSR (use Docker as apt-get upgrade is required)
- [ ] Upgrade models to generic English
- [ ] Add robocup grammar

MatthijsBurgh, 2018-10-06:
@ar13pit could you create seperate cards for each subtask? So people can assign themselves to each subtask.

ar13pit, 2018-10-06:
@MatthijsBurgh it's done.

MatthijsBurgh, 2018-12-22:
@ar13pit, could you check if some boxes can be ticked? And could you provide a small status update? Is end of January still reachable?

ar13pit, 2020-03-07:
Closed in favour of #1012

closed 2020-03-07 by ar13pit