tue-robotics/tue_robocup #663: Build isolated word recognition system

ar13pit, 2018-10-06:


alberth, 2018-10-09:
In the current speech recognition implementation (of Amigo), speech recognition only returns sentences that exist in the grammar file. To this end, the code does
- Load the grammar file into the parser
- Generate a large! file with all possible sentences (and it will grow larger, as we will spend less time on hardware-related issues, and thus more time on intelligent behavior).
- Give this file to speech recognition
- Convert this file for internal use in speech recognition
(some parts of this are perhaps due to limitations in the current speech package, I haven't looked into that)

On hearing a sentence, the current implementation returns a string of words, which are parsed again by the grammar, to get the semantics of it, which can be fed into the action server to make the robot do stuff.

It seems simpler if we can provide the parser itself to the speech recognition. It's a tree, it can tell you the next valid words at any point in a sentence directly.

alberth, 2018-10-09:
The system can recognize a spoken word without having a list  of available or "to be recognized" words at all? That's really nice.

MatthijsBurgh, 2018-10-09:
@alberth the question is, with the same accuracy?

ar13pit, 2018-10-09:
I think you guys are misinterpreting my comment. The system does need a list of words/grammar to give the output as words. What I mean here is we will make a really small isolated word recognition system (capable of recognizing limited words like only yes or no) only to make the ros wrappers and interface this system with the current software architecture.

Once this is done, the model can be replaced with something that has more phonemes which can recognize continuous speech.

closed 2018-11-21 by SamAlexandrov