tue-robotics/tue_robocup #669: Add robocup grammar

ar13pit, 2018-10-06:
This should be worked upon only after the closing of [#668](https://github.com/tue-robotics/tue_robocup/issues/668) as building of generic English speech recognition system should come first.

Also, the current grammar parser might have to be either rewritten or patched to work with the new speech recognition system.

ar13pit, 2018-10-09:
Reference to @alberth 's [comment](https://github.com/tue-robotics/tue_robocup/issues/663#issuecomment-428079823). 

Try to use the grammar parser directly with the system. As Kaldi works on FSTs, try to generate the graph using grammar parser.

SamAlexandrov, 2018-11-27:
As a first step, I'm looking at how to use [OpenFST](http://www.openfst.org/twiki/bin/view/FST/WebHome) to convert the grammar configuration file to G.fst model.

ar13pit, 2018-12-06:
Need to merge the grammar parser and some functions from dragonfly speech recognition to convert the grammar file `*.fcfg` into files required to make `G.fst`

closed 2019-04-07 by ar13pit