tue-robotics/tue_robocup #678: HERO laptop

MatthijsBurgh, 2018-10-23:


MatthijsBurgh, 2018-10-23:
Option:
https://www.coolblue.nl/product/813624/msi-gt63-8rg-035nl.html

ar13pit, 2018-10-30:
ASUS ROG
https://www.alternate.nl/ASUS/ROG-Zephyrus-GX501GI-EI005T-15-6-laptop/html/product/1465316?event=search

ar13pit, 2018-11-09:
@MatthijsBurgh By any chance can we take the advantage of Black Friday sale?

MatthijsBurgh, 2018-11-12:
MSI
https://www.alternate.nl/MSI/GT63VR-8RG-035NL-15-6-laptop/html/product/1430538?lk=23933

MatthijsBurgh, 2018-11-12:
The Asus laptop doesn't have an RJ45 port. Of course we could buy a dongle, but a build-in port is preferred. Downside of MSI, it has a higher resolution, but we could disable the gui, scale down resolution etc.

ar13pit, 2018-11-20:
@MatthijsBurgh Did you contact that company you were supposed to?

MatthijsBurgh, 2018-11-20:
They only have: https://bto.eu/product/bto-x-book-17cl879/
17 inch