tue-robotics/tue_robocup #685: Convert hardcoded joint goals to named poses in <robot>_bringup

LoyVanBeek, 2018-11-06:
This removes the need to use private method ```_send_joint_trajectory``` in many skills.

alberth, 2018-11-06:
Some are addressed in PR #686

MatthijsBurgh, 2019-03-17:
More work needed here?

alberth, 2019-03-19:
```` bash
$ ag _send_joint_trajectory
challenge_dishwasher/src/challenge_dishwasher/custom_place.py
35:            arm.resolve()._send_joint_trajectory([[-1.1, 0.044, 0.80, 0.297, 0.934, -0.95, 0.4]],
44:            arm.resolve()._send_joint_trajectory([[-0.92, 0.044, 0.80, 0.497, 0.934, -0.95, 0.4]],
70:            arm.resolve()._send_joint_trajectory([[-1.1, 0.044, 0.80, 0.297, 0.934, -0.95, 0.4]],

challenge_dishwasher/src/challenge_dishwasher/open_dishwasher.py
120:            robot.rightArm._send_joint_trajectory([[0, 0.2519052373022729913, 0.7746500794619434, 1.3944848321343395,
164:            robot.rightArm._send_joint_trajectory([[0, 0, 0, 0, 0, 0, 0]], timeout=rospy.Duration(0))
176:            robot.rightArm._send_joint_trajectory([[0, 1.57, 0, 0, 0, 0, 0]], timeout=rospy.Duration(0))
178:            robot.rightArm._send_joint_trajectory([[-1.5, 1.57, 0, 0, 0, 0, 0]], timeout=rospy.Duration(0))
180:            robot.rightArm._send_joint_trajectory([[-1.5, 0, 0, 0, 0, 0, 0]], timeout=rospy.Duration(0))
187:            robot.rightArm._send_joint_trajectory([[-0.8, 0, 0, 0, 0, 0, 0]], timeout=rospy.Duration(0))
218:            robot.rightArm._send_joint_trajectory([[-1.48, 0, 0, 0.5, 1.57, 0, -0.1]], timeout=rospy.Duration(0))
231:            robot.rightArm._send_joint_trajectory([[-1.4, 0, 0, 0.45, 1.57, 0.8, -0.1]], timeout=rospy.Duration(0))

challenge_final/src/challenge_final/open_door.py
117:            robot.rightArm._send_joint_trajectory([[0,0.7,0,1.3,1.57,0,0.4]], timeout=rospy.Duration(0))

challenge_storing_groceries/src/challenge_storing_groceries/open_door.py
147:        self.arm._send_joint_trajectory([[-0.101, 0.119, 0.200, 1.363, 0.230, 0.688, 0.280]])
148:        self.arm._send_joint_trajectory([[-0.574, 0.802, 0.691, 1.021, 0.750, 0.535, 0.378]])
149:        #self.arm._send_joint_trajectory([[-0.042, 0.720, 0.846, 1.039, 0.849, 0.534, 0.377]])
150:        self.arm._send_joint_trajectory([[-0.04446312115056816, 0.9908744970312501, 0.9101328212542089, 0.9240762995414635, 1.0067001691414637, 0.3949460482219828, 0.1621731308728449]])

challenge_presentation/src/challenge_presentation/presentation_hero.py
121:        function_list.append(partial(self.arm._send_joint_trajectory, [[0.01, 0.0, 0.0, -1.57, 0.0]]))
````
Apparently not done yet

MatthijsBurgh, 2021-01-27:
I think it is good to keep this one open.