tue-robotics/tue_robocup #697: [speech] Streaming microphone audio to ROS Publisher

DvanDinther, 2018-12-18:
This can be done with the audio_capture package from the [audio_common](https://github.com/ros-drivers/audio_common) package.

DvanDinther, 2018-12-18:
Currently working on trying to write a subscriber to this audio_capture publisher which can write the audio to an MP3 file, just for testing purposes. This can later be used in #666

alberth, 2019-01-03:
Why are you streaming audio through a publish/subscribe network? Seems a big waste of processing and network resources, and I don't see ever having more than 1 client.

ar13pit, 2019-01-03:
I completely agree. The reason why we chose this was to make the speech recognition system modular so that until the development is not complete and if we decide to switch to a different base library than Kaldi, the audio capture can easily be connected to that.

Also, there isn't much overhead as this is just a ros wrapper around gstreamer for audio capture.

closed 2019-02-20 by ar13pit