tue-robotics/tue_robocup #712: Find person in room broken (on master!!)

PetervDooren, 2019-02-05:
```
[INFO][/action_server][1549402216.216385]: 'Getting there!'
[INFO][/action_server][1549402216.229982]: State machine transitioning 'GET_PLAN':'goal_ok'-->'EXECUTE_PLAN'
[INFO][/action_server][1549402216.332356]: State machine terminating 'EXECUTE_PLAN':'succeeded':'arrived'
[INFO][/action_server][1549402216.333698]: State machine transitioning 'NAVIGATE':'arrived'-->'STOP_ANALYSIS_SUCCEED'
[INFO][/action_server][1549402216.437909]: State machine terminating 'STOP_ANALYSIS_SUCCEED':'done':'arrived'
[INFO][/action_server][1549402216.438512]: State machine transitioning 'NAVIGATE_TO_ROOM':'arrived'-->'FIND_PERSON'
[INFO][/action_server][1549402216.440281]: Trying to find someone
[INFO][/action_server][1549402216.441652]: 'someone, please look at me while I am looking for you'
[INFO][/action_server][1549402221.505715]: Creating subscriber
[INFO][/action_server][1549402221.510965]: lazy subscribe to /hsrb/head_rgbd_sensor/rgb/image_raw
[INFO][/action_server][1549402221.512127]: getting one image...
[INFO][/action_server][1549402221.512674]: self._camera_last_image: None
[INFO][/action_server][1549402221.577045]: len(self._camera_last_image): 921600
[INFO][/action_server][1549402221.578203]: got 921600 bytes of image data
[INFO][/action_server][1549402222.006609]: found 0 face(s) in the image
[INFO][/action_server][1549402222.007454]: best_detection = None
[INFO][/action_server][1549402227.030510]: getting one image...
[INFO][/action_server][1549402227.031848]: self._camera_last_image: None
[INFO][/action_server][1549402227.063703]: len(self._camera_last_image): 921600
[INFO][/action_server][1549402227.064240]: got 921600 bytes of image data
[INFO][/action_server][1549402227.408298]: found 1 face(s) in the image
[INFO][/action_server][1549402227.409903]: best_detection = categorical_distribution: 
  probabilities: []
  unknown_probability: 0.0
roi: 
  x_offset: 251
  y_offset: 85
  height: 72
  width: 62
  do_rectify: False
group_id: 0
[INFO][/action_server][1549402227.469250]: project_rois response: points: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1549402227
        nsecs: 190455198
      frame_id: "head_rgbd_sensor_rgb_frame"
    point: 
      x: -0.111222814485
      y: -0.306658449713
      z: 1.54499995708
Transforming roi to map
[WARN][/action_server][1549402227.476555]: Could not find someone
[INFO][/action_server][1549402232.487940]: getting one image...
[INFO][/action_server][1549402232.488464]: self._camera_last_image: None
[INFO][/action_server][1549402232.523858]: len(self._camera_last_image): 921600
[INFO][/action_server][1549402232.524638]: got 921600 bytes of image data
[INFO][/action_server][1549402232.919397]: found 1 face(s) in the image
[INFO][/action_server][1549402232.920285]: best_detection = categorical_distribution: 
  probabilities: []
  unknown_probability: 0.0
roi: 
  x_offset: 115
  y_offset: 75
  height: 86
  width: 74
  do_rectify: False
group_id: 0
[INFO][/action_server][1549402232.978223]: project_rois response: points: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1549402232
        nsecs: 753579140
      frame_id: "head_rgbd_sensor_rgb_frame"
    point: 
      x: -0.465437201742
      y: -0.302424584105
      z: 1.48199999332
Transforming roi to map
[WARN][/action_server][1549402232.989175]: Could not find someone
```

JosjaG, 2019-02-12:
This implies that the robot can't find a laser entity near the ROI, I think this has something to do with the height of Hero's laser, since that one is (much) lower than the one of Amigo (we've encountered this problem before). So I think there are many simple hacky solutions (and a few more time-consuming pretty solutions as well).

JosjaG, 2019-02-12:
Also what @LoyVanBeek and I looked at with https://github.com/tue-robotics/tue_robocup/blob/50dd8858f82427ee2ecb49ebcd2af8d8044bc793/robot_smach_states/src/robot_smach_states/navigation/follow_operator2_0.py#L410

LoyVanBeek, 2019-02-13:
There is a PR open for this: https://github.com/tue-robotics/tue_robocup/pull/718.

Needs to be rebased on master and a thorough review if this is the way to do this.

PetervDooren, 2019-04-02:
Not sure if this is caused by the same bug, but the error is different:
```
[INFO][/action_server][1554230290.843828]: got 921600 bytes of image data
[INFO][/action_server][1554230291.076481]: found 1 face(s) in the image
[INFO][/action_server][1554230291.076826]: get_best_face_recognition: recognitions = [categorical_distribution: 
  probabilities: []
  unknown_probability: 0.0
roi: 
  x_offset: 505
  y_offset: 256
  height: 126
  width: 105
  do_rectify: False
group_id: 0]
[INFO][/action_server][1554230291.077209]: Probability threshold 1.50
[INFO][/action_server][1554230291.077819]: 0: []
```

PetervDooren, 2019-04-02:
This error was caused by the variable "discard_other_labels" that is passed from the find action to the find smach state is set to True. This will cause the robot to always search for a known individual. The find action should be fixed so that it is possible to look for just anyone.

closed 2020-02-02 by LarsJanssenTUe