tue-robotics/tue_robocup #724: EdEntity.volumes broken

KevinDang74, 2019-02-26:
World_model_ed used the entity.data attribute to parse the volumes (e.g. `on_top_of`) of an entity from.
But for (some?) entities, this attribute is empty and has it's volumes encoded in a separate, dedicated attribute:

```yaml
[id: "trashbin"
type: ''
types: []
existence_probability: 1.0
last_update_time: 
  secs: 0
  nsecs:         0
creation_time: 
  secs: 0
  nsecs:         0
has_shape: True
has_pose: True
pose: 
  position: 
    x: 2.99237614478
    y: 3.62256310813
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: -0.336030446182
    w: 0.941851123712
data: '' # Empty
flags: []
convex_hull: 
  - 
    x: 0.15000000596
    y: 6.94901928e-06
    z: 0.0
  - much: 'more'
z_min: 0.0
z_max: 0.300000011921
volumes: #not empty!
  - 
    name: "on_top_of"
    subvolumes: 
      - 
        center_point: 
          header: 
            seq: 0
            stamp: 
              secs: 0
              nsecs:         0
            frame_id: "/trashbin"
          point: 
            x: 0.0
            y: 0.0
            z: 0.4
        geometry: 
          type: 1
          dimensions: [0.4, 0.2, 0.30000000000000004]]
```

MatthijsBurgh, 2019-02-26:
This is not broken. It is a new implementation....

LoyVanBeek, 2019-02-27:
I thought so... But it's not compatible with robot_skills.ED yet. What's the next step here?

MatthijsBurgh, 2019-02-27:
robot_skills is updated. The message has been updated and the function which converts the message to a Python instance. So the rest of the code doesn't need to be updated. I even did small check and couldn't find any code which could be broken.

So just be sure that whatever branch you are working on, is up-to-date with the github master, it should be working.

MatthijsBurgh, 2019-03-17:
@KevinDang74, issue resolved?

KevinDang74, 2019-03-19:
issue resolved.

closed 2019-03-19 by KevinDang74