tue-robotics/tue_robocup #725: Chrony failed on Hero backpack laptop

PetervDooren, 2019-03-09:
Chrony failed on `athome11`. 

1. Running `chronyc tracking` reported that the daemon could not be contacted. 
2. `systemctl status chrony` reports that it could not parse an `allow` line in its config. This line is `allow hero1.local`.
3. According to https://bugzilla.redhat.com/show_bug.cgi?id=1406445 a reason could be that the hostname was not resolvable. 
4. 1`systemctl restart chrony` fixed that, because by now the hostname can be resolved. 
 Potential final fix: Make `chrony.service` depend on `networking.target` in systemd?

MatthijsBurgh, 2019-03-17:
Still an issue on `HERO2`?

LoyVanBeek, 2019-03-18:
Not AFAIK (I wrote this issue with @PetervDooren). We put this on GitHub for future reference essentially.

PetervDooren, 2019-03-18:
We encountered this problem on athome11. Unless chrony.service works differently on hero2 I would expect to also see this there.

MatthijsBurgh, 2019-03-18:
I have not experienced it myself.

LoyVanBeek, 2019-03-18:
It depends on the ordering of startup I think, so timing and that can fail. We should add the networking service to the dependencies of the chrony service and not rely on timing but explicit ordering.

MatthijsBurgh, 2019-03-18:
Netwerk-online,but that would probably airtight.

PetervDooren, 2019-03-23:
I've tested this and it is indeed still a problem on hero2

MatthijsBurgh, 2019-03-23:
Indeed, I also experienced it. The problem is caused by the network not being online automatically.

closed 2019-12-15 by PetervDooren