tue-robotics/tue_robocup #727: Determine challenges we can do at German Open 2019

LoyVanBeek, 2019-03-09:
# Stage 1
## Housekeeper tasks
- [ ] clean-up
- [ ] gpsr
- [ ] storing-groceries
- [ ] serve-the-breakfast
- [ ] take-out-the-garbage

## Party Host tasks
- [ ] carry-my-luggage
- [ ] farewell
- [ ] find-my-mates
- [ ] receptionist
- [ ] serving-drinks

# Stage 2
## Housekeeper tasks
- [ ] clean-the-table
- [ ] egpsr
- [ ] find-my-disk
- [ ] set-the-table

## Party Host tasks
- [ ] hand-me-that
- [ ] stickler-for-the-rules
- [ ] smoothie-chef
- [ ] where-is-this

LoyVanBeek, 2019-03-13:
Also see https://github.com/RoboCupAtHome/RuleBook/issues/588

ar13pit, 2019-03-27:
@MatthijsBurgh @LoyVanBeek Which of these challenges require speech recognition? Because I am not assigned to anything in particular.

SamAlexandrov, 2019-03-27:
@ar13pit Maybe 'where-is-this' would be most suitable ;)

ar13pit, 2019-03-27:
@SamAlexandrov but that's stage 2

JosjaG, 2019-03-27:
@ar13pit gpsr definitely requires speech recognition

LoyVanBeek, 2019-03-27:
For the Receptionist task, we need to hear names and drinks AFAIK

LoyVanBeek, 2019-04-16:
Detectoing and classifying people is getting more important: https://github.com/tue-robotics/image_recognition/issues/87 
Challenge `receptionist`, `find_my_mates`, `serving_drinks` all depend on this.