tue-robotics/tue_robocup #730: Print Toyota tray attachment

MatthijsBurgh, 2019-03-12:


MatthijsBurgh, 2019-03-18:
See 
https://qa.hsr.io/en/question/1161/a-tray-that-is-held-by-the-hsr-using-the-dspl-restaurant-task/

MatthijsBurgh, 2019-03-18:
@SamAlexandrov could you do this?

MatthijsBurgh, 2019-03-25:
I already have somebody who can print these. @hvrooy, @LoyVanBeek @jlunenburg @JosjaG @LarsJanssenTUe @PetervDooren  Shall I print two of them?

LarsJanssenTUe, 2019-03-25:
Is it a problem to do it one by one? Because that is what I would do, so the design can be improved if necessary.

MatthijsBurgh, 2019-03-25:
We can do one by one, but I would like to take more than one to the German Open.

hvrooy, 2019-03-26:
I will have 'the University' print one item, and ask for a step file from the designer, so we may be able to change the tray interface for a tray of ours.

hvrooy, 2019-04-02:
Item has been printed, so it can be attached to the robot and tested. There is more than one way for HERO to hold the thing, though....

closed 2019-05-08 by hvrooy