tue-robotics/tue_robocup #740: Rename Hero arm to center

LarsJanssenTUe, 2019-03-26:
Currently it is named left arm, which we would like to change to a more general name. However this breaks currently in this piece of code:

```
    def configure(self):
        """
        This should be run at the end of the constructor of a child class.
        """
        self.arms = OrderedDict(left=self.leftArm, right=self.rightArm)  # ToDo: kind of ugly, why do we need this???

```
In robot.py

MatthijsBurgh, 2019-03-26:
I think I am the writer of these lines. The robot object requires/required to have an self.arms. I didn't want to put it in the robot specific constructor, because it shouldn't be there. There should just be`add_part` functions. But this solution is also very ugly and not generic.

alberth, 2019-04-02:
One way to go about this is ba136bcae0f66981a7ef088338811eab1330fcdc . You may also want to fix some Arm imports 731cd1fe1f97dd7342ade16c525876b78fa33d84

closed 2021-01-27 by PetervDooren