tue-robotics/tue_robocup #758: Recognize service does not assign probabilities

LarsJanssenTUe, 2019-04-23:
During the debugging some extra loggers were added (as can be seen) and eventually it was found that the recognize service:
 ``` 
self._recognize_srv = self.create_service_client('/' + robot_name + '/face_recognition/recognize', Recognize)
```
does not produce probabilities, therefore resulting in the following set of events:
```
[INFO][/state_machine][1556022471.898666]: getting one image...
[INFO][/state_machine][1556022471.898999]: self._camera_last_image: None
[INFO][/state_machine][1556022471.933521]: len(self._camera_last_image): 921600
[INFO][/state_machine][1556022471.933880]: got 921600 bytes of image data
[INFO][/hero/face_recognition/face_recognition][1556022472.172789]: Face recognitions: [RecognizedFace(roi=(474, 189, 127, 149), l2_distances=[])]
[INFO][/state_machine][1556022472.177438]: recognize service recognitions: 
  - 
    categorical_distribution: 
      probabilities: []
      unknown_probability: 0.0
    roi: 
      x_offset: 474
      y_offset: 189
      height: 149
      width: 127
      do_rectify: False
    group_id: 0
[INFO][/state_machine][1556022472.177817]: found 1 face(s) in the image
[INFO][/state_machine][1556022472.178329]: Probability threshold 1.50
[INFO][/state_machine][1556022472.178566]: 0: []
[INFO][/state_machine][1556022472.178850]: raw_detections = [categorical_distribution: 
  probabilities: []
  unknown_probability: 0.0
roi: 
  x_offset: 474
  y_offset: 189
  height: 149
  width: 127
  do_rectify: False
group_id: 0]
[INFO][/state_machine][1556022472.179074]: best_detection = None

```

PetervDooren, 2021-01-27:
@LarsJanssenTUe Can this be closed?

LarsJanssenTUe, 2021-02-02:
Well the issue is probably not solved (as far as I know), so face recognition probably does not work. In this issue you get some information about what is wrong, that being said I do not think this issue has priority so you can close it if you want.