tue-robotics/tue_robocup #840: Receptionist Test Logs

MatthijsBurgh, 2019-05-22:


MatthijsBurgh, 2019-05-22:
```
[ERROR][/state_machine][1558532918.197395]: InvalidUserCodeError: Could not execute state 'WAIT_FOR_GUEST' of type '<robot_smach_states.human_interaction.human_interaction.WaitForPersonInFront object at 0x7eff9408f250>': Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/human_interaction/human_interaction.py", line 243, in execute
    success, found_people_ids = self.robot.ed.detect_people(*image_data)
  File "/home/amigo/ros/kinetic/system/src/robot_skills/src/robot_skills/world_model_ed.py", line 503, in detect_people
    result = self._ed_detect_people_srv(rgb, depth, cam_info)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/hero/ed/people_recognition/detect_people] responded with an error: 

[ERROR][/state_machine][1558532918.197860]: State 'LEARN_GUEST_1' failed to execute.
[ERROR][/state_machine][1558532918.198216]: Robot exited with <class 'smach.exceptions.InvalidUserCodeError'>,Could not execute state 'WAIT_FOR_GUEST' of type '<robot_smach_states.human_interaction.human_interaction.WaitForPersonInFront object at 0x7eff9408f250>': Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/human_interaction/human_interaction.py", line 243, in execute
    success, found_people_ids = self.robot.ed.detect_people(*image_data)
  File "/home/amigo/ros/kinetic/system/src/robot_skills/src/robot_skills/world_model_ed.py", line 503, in detect_people
    result = self._ed_detect_people_srv(rgb, depth, cam_info)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/hero/ed/people_recognition/detect_people] responded with an error: 
,<traceback object at 0x7eff9407d098>
```

LoyVanBeek, 2019-05-23:
This test revolves around dealing with people. If we can't do that, there is no point in continuing. But a nice warning message and retry would be nice

closed 2019-08-31 by LarsJanssenTUe