tue-robotics/tue_robocup #842: Where is this Test logs

LarsJanssenTUe, 2019-05-23:


PetervDooren, 2019-05-29:
**Challenge crashes at startup**
running is:
-hero-start on hero1
-hero-free-mode on hero2
challenge itself:
in ~/ros/kinetic/system/src/challenge_where_is_this/scripts run ./challenge_where_is_this hero

Running challenge_where_is_this throws the following error

[ERROR][/where_is_this][1559123443.800201]: Error raised during SMACH container construction: 
Traceback (most recent call last):

  File "/home/amigo/ros/kinetic/system/src/challenge_where_is_this/src/challenge_where_is_this/inform_machine.py", line 116, in __init__
    states.HearOptionsExtra(robot, self.spec_des, self.answer_des.writeable, rospy.Duration(15)),

  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/human_interaction/human_interaction.py", line 150, in __init__
    ds.check_resolve_type(choices_designator, dict)

  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/designators/checks.py", line 45, in check_resolve_type
    raise TypeError("{0} resolves to {1} but should resolve to one of {2}".format(designator, designator.resolve_type, allowed_types))

TypeError: VariableWriter(<class 'robot_smach_states.util.designators.core.VariableDesignator'>(..., name=None)) resolves to <class 'hmi.common.HMIResult'> but should resolve to one of (<type 'dict'>,)

[ERROR][/where_is_this][1559123443.800590]: Error raised during SMACH container construction: 
Traceback (most recent call last):

  File "/home/amigo/ros/kinetic/system/src/challenge_where_is_this/src/challenge_where_is_this/where_is_this.py", line 22, in __init__
    single_item = InformMachine(robot)

  File "/home/amigo/ros/kinetic/system/src/challenge_where_is_this/src/challenge_where_is_this/inform_machine.py", line 116, in __init__
    states.HearOptionsExtra(robot, self.spec_des, self.answer_des.writeable, rospy.Duration(15)),

  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/human_interaction/human_interaction.py", line 150, in __init__
    ds.check_resolve_type(choices_designator, dict)

  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/designators/checks.py", line 45, in check_resolve_type
    raise TypeError("{0} resolves to {1} but should resolve to one of {2}".format(designator, designator.resolve_type, allowed_types))

TypeError: VariableWriter(<class 'robot_smach_states.util.designators.core.VariableDesignator'>(..., name=None)) resolves to <class 'hmi.common.HMIResult'> but should resolve to one of (<type 'dict'>,)

[ERROR][/where_is_this][1559123443.800881]: Robot exited with <type 'exceptions.TypeError'>,VariableWriter(<class 'robot_smach_states.util.designators.core.VariableDesignator'>(..., name=None)) resolves to <class 'hmi.common.HMIResult'> but should resolve to one of (<type 'dict'>,),<traceback object at 0x7fc5dc0d6a70>
[INFO][/where_is_this][1559123443.801141]: left arm cancelling all goals on all arm-related ACs on close
[INFO][/where_is_this][1559123443.801692]: Torso cancelling all goals on close
Traceback (most recent call last):
  File "./challenge_where_is_this", line 11, in <module>
    startup(WhereIsThis)
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/startup.py", line 70, in startup
    executioner = statemachine_creator(robot)
  File "/home/amigo/ros/kinetic/system/src/challenge_where_is_this/src/challenge_where_is_this/where_is_this.py", line 22, in __init__
    single_item = InformMachine(robot)
  File "/home/amigo/ros/kinetic/system/src/challenge_where_is_this/src/challenge_where_is_this/inform_machine.py", line 116, in __init__
    states.HearOptionsExtra(robot, self.spec_des, self.answer_des.writeable, rospy.Duration(15)),
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/human_interaction/human_interaction.py", line 150, in __init__
    ds.check_resolve_type(choices_designator, dict)
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/designators/checks.py", line 45, in check_resolve_type
    raise TypeError("{0} resolves to {1} but should resolve to one of {2}".format(designator, designator.resolve_type, allowed_types))
TypeError: VariableWriter(<class 'robot_smach_states.util.designators.core.VariableDesignator'>(..., name=None)) resolves to <class 'hmi.common.HMIResult'> but should resolve to one of (<type 'dict'>,)

MatthijsBurgh, 2019-05-29:
@PetervDooren, which `tue_robocup` branch did you use?

PetervDooren, 2019-05-29:
Rgo2019 challenge where is this
I heard from lars that you fixed this at some point

MatthijsBurgh, 2019-05-29:
You should test on the RGO2019 branch.

ar13pit, 2019-06-13:
1. Openpose does not automatically start on Hero2, also not with free mode.

2. @jlunenburg :
```bash
[~/ros/kinetic/system/src/robot_smach_states/test] (challenge-where-is-this-guidance $=)
amigo@hero2 $ python test_guidance.py hero dinner_table
[INFO][/test_guidance][1560444333.100822]: Constructing robot hero
[ERROR][/test_guidance][1560444340.614412]: /hero/head_rgbd_sensor/project_2d_to_3d not connected timely
[ERROR][/test_guidance][1560444341.488885]: /hero/emergency_switch not connected timely
[ERROR][/test_guidance][1560444342.318260]: /hero/hmi/dragonfly_speech_recognition/restart_node not connected timely
[ERROR][/test_guidance][1560444342.819244]: /hero/torso/measurements not connected timely
[WARN][/test_guidance][1560444342.820428]: Not all hardware operational: ['head', 'base', 'torso']
[INFO][/test_guidance][1560444342.821205]: State machine starting in initial state 'GET_PLAN' with userdata: 
	[]
[INFO][/test_guidance][1560444343.099224]: 'Affirmative!'
[INFO][/test_guidance][1560444343.167555]: State machine transitioning 'GET_PLAN':'goal_ok'-->'EXECUTE_PLAN'
[INFO][/test_guidance][1560444346.676755]: State machine transitioning 'EXECUTE_PLAN':'blocked'-->'PLAN_BLOCKED'
[INFO][/test_guidance][1560444346.677019]: Plan blocked
[INFO][/test_guidance][1560444348.679607]: Plan free again
[INFO][/test_guidance][1560444348.679900]: State machine transitioning 'PLAN_BLOCKED':'free'-->'EXECUTE_PLAN'
[INFO][/test_guidance][1560444356.150035]: Received rgb, depth, cam_info
[ERROR][/test_guidance][1560444358.446548]: InvalidUserCodeError: Could not execute state 'EXECUTE_PLAN' of type '<robot_smach_states.navigation.guidance.ExecutePlanGuidance object at 0x7f02241e08d0>': Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/navigation/guidance.py", line 99, in execute
    if not self._check_operator():
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/navigation/guidance.py", line 113, in _check_operator
    return detect_operator_behind_robot(self.robot)  # , self._follow_distance, self._operator_radius_threshold)
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/navigation/guidance.py", line 37, in detect_operator_behind_robot
    base_pose = robot.pose.get_location()
AttributeError: 'Hero' object has no attribute 'pose'

Traceback (most recent call last):
  File "test_guidance.py", line 51, in <module>
    s.execute()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/smach/state_machine.py", line 359, in execute
    container_outcome = self._update_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/smach/state_machine.py", line 254, in _update_once
    + traceback.format_exc())
smach.exceptions.InvalidUserCodeError: Could not execute state 'EXECUTE_PLAN' of type '<robot_smach_states.navigation.guidance.ExecutePlanGuidance object at 0x7f02241e08d0>': Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/navigation/guidance.py", line 99, in execute
    if not self._check_operator():
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/navigation/guidance.py", line 113, in _check_operator
    return detect_operator_behind_robot(self.robot)  # , self._follow_distance, self._operator_radius_threshold)
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/navigation/guidance.py", line 37, in detect_operator_behind_robot
    base_pose = robot.pose.get_location()
AttributeError: 'Hero' object has no attribute 'pose'
```

MatthijsBurgh, 2019-06-13:
Openpose does start automatically, It starts too early, therefore it fails.

jlunenburg, 2019-06-18:
Fixed in 40228a965bae6dec6363c405c94e44a7c3523447 , still needs to be tested on the real robot

PetervDooren, 2019-06-18:
New feature/bug

```

[INFO][/test_guidance][1560881908.235466]: Received rgb, depth, cam_info
[ERROR][/test_guidance][1560881909.320299]: Cannot instantiate PersonProperties from age: 36
emotion: 
gender: 0
gender_confidence: 0.789691
name: 
posture: 
reliability: 0
shirt_colors: [light blue]
tagnames: []
tags: []
header: 
    frame_id: head_rgbd_sensor_rgb_frame
    seq: 535
    stamp: 
        nsec: 118855946
        sec: 1560881907
pointing_pose: 
    orientation: 
        w: 0
        x: 0
        y: 0
        z: 0
    position: 
        x: 0
        y: 0
        z: 0
position: 
    x: 0.0334777
    y: 0.171243
    z: 1.34775
velocity: 
    x: 0
    y: 0
    z: 0

```

closed 2019-08-31 by LarsJanssenTUe