tue-robotics/tue_robocup #845: Cleanup Test logs

PetervDooren, 2019-05-29:


PetervDooren, 2019-05-29:
Challenge cleanup breaks tue-make. It throws the following error

Errors     << challenge_cleanup:check /home/amigo/ros/kinetic/system/logs/challenge_cleanup/build.check.015.log   
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_python_setup.cmake:31 (message):
  catkin_python_setup() called without 'setup.py' in project folder '
  /home/amigo/ros/kinetic/system/src/challenge_cleanup'
Call Stack (most recent call first):
  CMakeLists.txt:10 (catkin_python_setup)

make: *** [cmake_check_build_system] Error 1
cd /home/amigo/ros/kinetic/system/build/challenge_cleanup; catkin build --get-env challenge_cleanup | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

PetervDooren, 2019-05-29:
Challenge fails to start using 'hero-challenge-cleanup', throws the following error:

ERROR: Unknown node(s):
 * /state_machine
... logging to /home/amigo/.ros/log/20190529-090840_96424ae4-81e0-11e9-a66c-00306444dcd8/roslaunch-hero2-4163.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

unused args [robot_name] for include of [/home/amigo/ros/kinetic/system/src/robot_launch_files/launch/world_modeling/image_recognition.launch]
The traceback for the exception was written to the log file

Can be caused by the fact that cleanup.py still expects a rosparam "robot_name" and "skip"

PetervDooren, 2019-05-29:
> Challenge cleanup breaks tue-make. It throws the following error
> 
> Errors << challenge_cleanup:check /home/amigo/ros/kinetic/system/logs/challenge_cleanup/build.check.015.log
> CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_python_setup.cmake:31 (message):
> catkin_python_setup() called without 'setup.py' in project folder '
> /home/amigo/ros/kinetic/system/src/challenge_cleanup'
> Call Stack (most recent call first):
> CMakeLists.txt:10 (catkin_python_setup)
> 
> make: *** [cmake_check_build_system] Error 1
> cd /home/amigo/ros/kinetic/system/build/challenge_cleanup; catkin build --get-env challenge_cleanup | catkin env -si /usr/bin/make cmake_check_build_system; cd -
> 
> This would suggest that a setup.py file is required in the challenge_cleanup folder

This was fixed by @MatthijsBurgh.

PetervDooren, 2019-06-11:
Testing the cleanup challenge resulted in the following error.

After specifying which room to clean and confirming the selection. The robot gave this

[INFO][/state_machine][1560261872.359695]: 'All I want to do is clean this mess up!'
[INFO][/state_machine][1560261874.316340]: State machine transitioning 'SAY_START_CHALLENGE':'spoken'-->'ITERATE_NEXT_LOC'
[INFO][/state_machine][1560261874.316612]: State machine transitioning 'ITERATE_NEXT_LOC':'next'-->'INSPECT'
[INFO][/state_machine][1560261874.316959]: State machine starting in initial state 'ITERATE_NEXT_AREA' with userdata: 
	[]
[INFO][/state_machine][1560261874.317160]: State machine transitioning 'ITERATE_NEXT_AREA':'next'-->'INSPECT'
[INFO][/state_machine][1560261874.317506]: State machine starting in initial state 'START_ANALYSIS' with userdata: 
	[]
[INFO][/state_machine][1560261874.418087]: State machine transitioning 'START_ANALYSIS':'done'-->'NAVIGATE'
[INFO][/state_machine][1560261874.418412]: State machine starting in initial state 'GET_PLAN' with userdata: 
	[]
[ERROR][/state_machine][1560261874.521361]: ERROR: robot.ed.get_entities(id=Designator(resolve_type=<type 'str'>, name=location_id), type=, center_point=VectorStamped([           0,           0,           0] @ /map), radius=0)
[ERROR][/state_machine][1560261874.521641]: L____> [field id must be of type str]
[ERROR][/state_machine][1560261874.521823]: No entities found in Designator(resolve_type=<class 'robot_skills.util.entity.Entity'>, name=e_des)
[ERROR][/state_machine][1560261874.522088]: Designator Designator(resolve_type=<class 'robot_skills.util.entity.Entity'>, name=e_des) in entity_designator_area_name_map resolved to None.
[ERROR][/state_machine][1560261874.522505]: ERROR: robot.ed.get_entities(id=Designator(resolve_type=<type 'str'>, name=room_id), type=, center_point=VectorStamped([           0,           0,           0] @ /map), radius=0)
[ERROR][/state_machine][1560261874.522701]: L____> [field id must be of type str]
[ERROR][/state_machine][1560261874.522988]: Designator EntityByIdDesignator(id=Designator(resolve_type=<type 'str'>, name=room_id), name=room_des) in entity_designator_area_name_map resolved to None.
[ERROR][/state_machine][1560261874.523167]: At least 1 designator in self.entity_designator_area_name_map failed
[WARN][/state_machine][1560261874.523333]: Invalid constraints given to getPlan(). constraint: None
[INFO][/state_machine][1560261874.523502]: State machine terminating 'GET_PLAN':'goal_not_defined':'goal_not_defined'

JosjaG, 2019-06-11:
@PetervDooren Please use triple quotes to format errors (this greatly improves readability :) ).

closed 2019-08-31 by LarsJanssenTUe