tue-robotics/tue_robocup #849: Tune self._ed_classify_srv(ids=ids, unknown_probability=0.3)

LoyVanBeek, 2019-06-01:
As per https://github.com/tue-robotics/tue_robocup/pull/837#discussion_r286377169

PetervDooren, 2019-06-25:
Changed it so that unknown probability is a property of the ed robot part. If a challenge requires a different unknown probability they can add in their challenge
```
robot.ed.set_unknown_probability(0.XX)
```

MatthijsBurgh, 2019-06-25:
Why is it a property of the ed part? It could be a argument of the function call? When it has a default value, we don't need to change all the code which calls this function.

LoyVanBeek, 2019-06-25:
I wouldn't use a setter-function but a `@property`:
```python
@property
unknown_probability(self):
    return self._unknown_probability

@unknown_probability.setter
unknown_probability(self, value):
    self._unknown_probability = value

@MatthijsBurgh This allows you to set this once for your challenge without the need to bubble this everywhere throughout your code, in challenges, designators, states etc.

closed 2021-02-02 by PetervDooren