tue-robotics/tue_robocup #859: Weird grammar memory

JosjaG, 2019-06-12:
The first time when the challenge has just started, everything goes well, but at the second command some kind of memory seems to have built up in the grammar. Instead of listening for the difference between `yes` and `no` the robot immediately understand something in the lines of the previously received command. 
``` 
[INFO][/state_machine][1560330655.511453]: 'I heard bring me the fanta from the dinner table, is this correct?'
[INFO][/state_machine][1560330659.588011]: Question: , grammar: T -> yes | no
[INFO][/state_machine][1560330659.588418]: Example: 'no'
[INFO][/state_machine][1560330660.096768]: Robot heard 'deliver fanta from the dinner_table to the plant' False
```
When the challenge is restarted, the first command goes ok again and at the second command the same problem occurs.

MatthijsBurgh, 2019-06-12:
Probably the speech bug: https://github.com/tue-robotics/dragonfly_speech_recognition/issues/14

MatthijsBurgh, 2019-06-12:
@jlunenburg @reinzor @LoyVanBeek Does this sound as the speech bug to you?

MatthijsBurgh, 2019-08-12:
This is the speech bug. Therefore closing this duplicate issue.

closed 2019-08-12 by MatthijsBurgh