tue-robotics/tue_robocup #864: Focus after Sydney

LoyVanBeek, 2019-06-25:
After the RWC2019 in Sydney, I think we should focus on stability. 
The reason for using HERO was to have stability of the hardware and software so we could focus on higher level challenges. This has so far not yet worked out. 
Having a more stable robot would potentially also allow the demo team to use HERO, as it's also much easier to transport than AMIGO. 

As @jlunenburg suggested, maybe we should split tue_robocup in two: challenges and API below it (robot_smach_states + robot_skills). Both can be held to different standards in terms of code quality and review.

I think it will also be worthwhile to set up more tests to ensure nothing breaks. At @mojin-robotics, I set up a lot of scriptable action servers and services that mock what e.g. robot_skills would be taking to. This allows me to test our applications logic level without even a simulator. 

Same goes for ED: it would be nice to just play a bag file and compare the generated WM from that to what we expect. 

I'm not sure how to to improve stability on the lower levels, but we definitely need this.

ar13pit, 2019-06-25:
I agree. Moving robot_smach_states and skills to new repo will ensure better maintainability of these core libraries. We may consider having a different high level repo for things like issues across the organisation and Kanban boards.

Also something to consider is the target system we use in `tue-env` for `ros` packages. If you try to manually setup a package by looking into the contents of package.xml for dependencies, a lot of our packages will break. It is good to have targets but package installation must be independent of it.

MatthijsBurgh, 2019-06-25:
@arpit you mean the dependencies in the package.xml are not up-to-date?

ar13pit, 2019-06-25:
> @ar13pit you mean the dependencies in the package.xml are not up-to-date?

In a way I mean that. What I'm trying to stress upon is that the dependencies of packages are not well defined in the packages, whether package.xml or not. The front-end packages definitely have this issue (for example the bringup packages). It is rare with backend packages but still there. The problem can easily be detected and solved by setting up unit tests.

We need to make good unit tests mandatory in each package.

LoyVanBeek, 2019-06-26:
Regarding `tue-env`, I think is has 1 core flaw: the dependencies and installation procedure for a package should be defined in the package itself, not in a separate repository. A .deb-package has a preinst,postinst script etc, that could be a place for some configurations but .debs can also put config files in the correct place automatically.

LoyVanBeek, 2019-06-26:
I would also find it very handy to just load a recorded world model and run test queries against it.

Rayman, 2019-06-26:
I've made an effort to put all the python code in one repo so it's much easier to commit atomic changes. Please don't undo this. When you  change the API, you should change all users in the same commit.

alberth, 2019-06-27:
Some steps you can take:
1. Stop seeing the software as a single entity, and think of it as a collection of (theoretically completely) independent packages with well-defined interfaces between them.
2. To handle gradual changes in the API, allow support of several interfaces to a single package with semantic versioning.
3. For management of the interfaces, any package should probably state what other package interfaces it needs with a version.
4. To enforce a package doesn't break its promises, rigorously verify API usage.

Some comments:
- ad 1: We tried that last year, I haven't seen any package diagram or interface description. Even if that was done, I doubt it would be maintained.
- ad 2: multi-package updates scale badly (even in a single repo imho), which would not be needed any more in this way. The procedure to change an API becomes then:
  1. Update the package providing the API, and add a new interface with a new version.
  2. Update all its users for the new API interface (doesn't need to be done at the same time).
  3. Drop the old API interface from the providing package (can be done only after all users have been updated).

However
- ad 3: You definitely need to know what API interface versions are used throughout the system in some way
- ad 4: And you definitely need to know that API interface usage is actually current. Python being pretty much a total disaster in keeping its promises implies something is needed here. Switching to another language could also work, eg a compiled Java would catch lots of errors at compile time rather than stumbling upon them in the middle of a robot test.

Of course, any system is not going to work as long as people just want to hack new features. Systems that are not in code are not created or not updated with code changes, systems in code are by-passed, or simply completely disabled hours after hitting master.

Finally,

As an example of what could happen, see https://fei.edu.br/rcs/2014/RegularPapers/robocupsymposium2014_submission_23.pdf  (Please judge it only on their their ideas, and chosen conceptual path, and the change of how they changed their vision of the software now. Likely their implementation won't work for us, but a different implementation based on the same ideas should be possible.)

closed 2020-03-31 by LarsJanssenTUe