tue-robotics/tue_robocup #865: Designators do not resolve to the same entity

PetervDooren, 2019-06-25:
In many challenges and robot smach states we assume that a designator will always resolve to the same entity. However there's no mechanism that ensures that this is the case. 

@jlunenburg suggested that we either assert that is a locking designator and lock it (not so convenient for the user) or use userdata to this end (easier for the end-user, less reusable).

This is an issue for after Sydney.

MatthijsBurgh, 2019-06-25:
Lock that shit

LoyVanBeek, 2019-06-30:
There's a bunch of states for locking and unlocking a designator. 
It's not convenient indeed. 

What could be handy is some syntax like this:
```python
with locked(some_designator) as locked:
    smach.StateMachine.add("FOO" , 
                           Bar(robot, locked), 
                           transitions={...})
```

closed 2020-02-02 by LarsJanssenTUe