tue-robotics/tue_robocup #869: Generalize Robot class better

ar13pit, 2019-06-30:
Amigo, Hero, Sergio classes have repeated statements in the `__init__` method. Instead of adding parts to the base class Robot, we can generalize it better by making most of these parts as properties of the general Robot class and selectively activating the required ones for each robot.

Or something even better would be an intermediate class that has all these parts added and we inherit Amigo, Hero and Sergio from this class.

LoyVanBeek, 2019-07-01:
That add more indirection. We had something similar earlier but converted to the current composing of each specific robot from the actual parts.

alberth, 2019-07-02:
Moving such code to some common function reduces line count indeed, but makes customizing how a part is instantiated for one of the robots more complicated.

In all, this code isn't changing much or executed often, so having less than optimal line count isn't terribly important.

closed 2020-02-02 by LarsJanssenTUe