tue-robotics/tue_robocup #877: ED Plugins in one master repo

ar13pit, 2019-07-11:
We have a lot of ED plugins and all have separate repos. This is painful in terms of maintenance and locating plugins.

We can have a master repo called ed_plugins with a meta package and all the plugins as sub dirs/packages of this master repo.

MatthijsBurgh, 2019-07-12:
I don't agree (yet). So please add extra arguments

LarsJanssenTUe, 2020-03-31:
@ar13pit ping! Close or continue discussion please

ar13pit, 2020-03-31:
@MatthijsBurgh I'm still for putting all the plugins in one repo as a multiple packages. It will be better maintained.

MatthijsBurgh, 2020-03-31:
I am still not convinced...

closed 2021-01-27 by PetervDooren

MatthijsBurgh, 2021-01-27:
Yes, it is closed 🥳

ar13pit, 2021-01-29:
Haha well neither of us agreed upon anything. So best to leave ED as it is till it lives :)