tue-robotics/tue_robocup #879: HMI bug

ar13pit, 2019-07-11:
1. New request must not to be sent to the clients until a response from the previous is received.
2. Before sending a new request, a preempt request must be sent to all active clients processing the previous request.
3. The clients must send a status of the preempt request and if no such request is received by the server, then respawn those clients.

If this is not done then grammar memory is fucked!

MatthijsBurgh, 2019-07-12:
These are all action clients, so should be possible. I am not so sure about the respawning. Maybe some action id checks might help.

Rayman, 2019-07-12:
I think in principle the first 2 points you mentioned are implemened [here](https://github.com/tue-robotics/hmi/blob/master/hmi/scripts/multi_client), but there could be a bug. The main loop is single threaded. All requests are handled before a new request is received.

Idea's
- After clients are cancelled, there is no confirmation if the cancellation succeeded (`wait_for_result`)
- ???

ar13pit, 2019-07-12:
Still needs testing though, just to be sure. And then we need to put docs in it so that we don't end up looking into this again.

LarsJanssenTUe, 2020-03-31:
@ar13pit what do you want to do with this?

ar13pit, 2020-03-31:
This will be checked in the coming weeks when we are testing the grammar construction for speech recognition.