tue-robotics/tue_robocup #882: Dragonfly (the infamous speech bug)

ar13pit, 2019-07-11:
Apart from the issues of HMI server, the bug is in our server interface of dragonfly speech recognition. The preempts are not being done correctly and a new request is overwriting the grammar before completion of the existing one.

MatthijsBurgh, 2019-07-12:
https://github.com/tue-robotics/dragonfly_speech_recognition/issues/14

Rayman, 2019-07-12:
The famous dragonfly bug where grammars can't be unloaded :(

MatthijsBurgh, 2019-07-12:
> The famous dragonfly bug where grammars can't be unloaded :(

Is it a bug in dragonfly itself? We could switch to a new version, https://github.com/tue-robotics/dragonfly_speech_recognition/issues/26, this could solve it, if it is in dragonfly.

Rayman, 2019-07-12:
Dragonfly has COM bindings to the underlying windows speech service. Even when you unload the grammers, you still receive callbacks when things are heared by the WSR engine. Maybe this behavior is fixed in the new version?

Rayman, 2019-07-12:
Maybe using a different engine, kaldi or dragon fixes it.

ar13pit, 2019-07-12:
We could try with the new dragonfly and see if it is fixed.

DNS maybe a good alternative.

For Kaldi, we won't use dragonfly as we have already built our custom grammar loading and unloading functionality and it works. The only thing remaining in it is our customized model inference pipeline which needs a bit of work.