tue-robotics/tue_robocup #884: Auto doc pages generation

ar13pit, 2019-07-12:
I think its high time we do something about documentation. Currently we have the following major documentation sources:
1. tue_robocup (WIKI page)
2. tue-robotics.github.io (WIKI page)
3. tue-robotics.github.io (Website) (Only SW architecture)

**What's needed:**
1. _A central place for documentation of all of our repos_:
I have two options in mind and would like all opinions on this. The first is a `docs` subdomain in `techunited.nl` and second is `tue-robotics.github.io`

2. _These docs must be auto generated from the doc strings of our software and no code should be allowed to be merged to master without proper documentation!_

3. _The structure of this portal should be such that these docs become our team bible and every new team member (initiate :P) just needs to go through this to know what we have and not spend time trying to understand minor details from various team members._

@LoyVanBeek @jlunenburg I'm sure you guys have more to add onto this and ideas about how to get this done.

LoyVanBeek, 2019-07-12:
I've been busy setting up https://readthedocs.org/ for smach states and robot_skills, but never finished that.

ar13pit, 2019-07-12:
AFAIK, with `RTD` each repository ends up having a separate url for the docs (Correct me if I am wrong).

What we can do is add some tooling to `tue-env` to auto generate all of our docs in a Docker container and publish it onto a single domain through `RTD`. Also setup hooks to trigger this build if the `master` branch in any of our repos has a new commit.

I can try setting this up unless there are better ideas :smile:

LoyVanBeek, 2019-07-12:
Go ahead, I'd say. Perhaps adding a https://docs.readthedocs.io/en/stable/config-file/v2.html already helps

ar13pit, 2019-07-12:
I'll at least set up a framework where all the other docs can be added.

ar13pit, 2019-07-12:
Ref 1. https://github.com/steveklabnik/automatically_update_github_pages_with_travis_example

LarsJanssenTUe, 2020-03-31:
@MatthijsBurgh I think this can be closed?

ar13pit, 2020-03-31:
Yeah.

closed 2020-03-31 by ar13pit