tue-robotics/tue_robocup #885: RWC2019 aftermath

MatthijsBurgh, 2019-08-12:
- [ ] action_server
- [ ] amigo_bringup
- [ ] amigo_description
- [ ] amigo_hardware
- [ ] amigo_moveit_config
- [ ] amigo_msgs
- [ ] amigo_ref_interpolator
- [ ] ar_marker_detector
- [ ] battery_manager
- [ ] cb_base_navigation
- [ ] cb_planner_msgs_srvs
- [ ] code_profiler
- [ ] control_msgs
- [ ] conversation_engine
- [ ] dashboard
- [ ] dialogflow_ros
- [ ] dragonfly_speech_recognition
- [ ] dynamixel_jointstate_conversion
- [ ] dynamixel_motor
- [ ] dynamixel_ros_control
- [ ] ed
- [ ] ed_gui_server
- [ ] ed_localization
- [ ] ed_moveit
- [ ] ed_msgs
- [ ] ed_navigation
- [ ] ed_object_models
- [ ] ed_people_recognition
- [ ] ed_perception
- [ ] ed_perception_models
- [ ] ed_robocup
- [ ] ed_rviz_plugins
- [ ] ed_sensor_integration
- [ ] ed_tutorials
- [ ] ed_wire
- [ ] emc-env
- [ ] emc_simulator
- [ ] emc_system
- [ ] emergency_speakup
- [ ] ethercat_interface
- [ ] fast_simulator
- [ ] fast_simulator_data
- [ ] geolib2
- [ ] GPSRCmdGen
- [ ] gitim
- [ ] grammar_parser
- [ ] gripper_detector
- [ ] head_ref
- [ ] head_ref_ed_client
- [ ] hero-dashboard
- [ ] hero-display
- [ ] hero_bridge
- [ ] hero_bringup
- [ ] hero_description
- [ ] hmi
- [ ] hmi_telegram
- [ ] image_recognition
- [ ] image_recognition-release
- [ ] image_recognition_wire
- [ ] kaldi
- [ ] kaldi_srilm
- [ ] kinect_driver
- [ ] matrix_creator_ros
- [ ] multirobot_communication
- [ ] navigation
- [ ] nl_robot_console
- [ ] node_alive
- [ ] people_recognition
- [ ] pico_base_controller
- [ ] pico_bringup
- [ ] pico_talk
- [ ] qr_detector_zbar
- [ ] r5cop_demo
- [ ] rgb_lights_manager
- [ ] rgbd
- [ ] rgbd_msgs
- [ ] robot-api
- [ ] robot_launch_files
- [ ] rtt
- [ ] rtt_control_components
- [ ] rtt_soem
- [ ] rtt_tue_msgs
- [ ] scl_filters
- [ ] sergio_bringup
- [ ] sergio_control
- [ ] sergio_description
- [ ] sergio_gripper_moveit_config
- [ ] sergio_hardware
- [ ] sergio_moveit_config
- [ ] sergio_torso_calibration
- [ ] simulink_models
- [ ] slack_ros
- [ ] snowboy_ros
- [ ] soem
- [ ] speech_recognition
- [ ] telegram_ros
- [ ] teleop_twist_web
- [ ] text_to_speech
- [ ] tf_server
- [ ] tue-env
- [ ] tue-env-targets
- [ ] tue-robotics.github.io
- [ ] tue_config
- [ ] tue_docs
- [ ] tue_ethercat_drivers
- [ ] tue_filesystem
- [ ] tue_manipulation
- [ ] tue_manipulation_msgs
- [ ] tue_mobile_ui
- [ ] tue_msgs
- [ ] tue_robocup
- [ ] tue_serialization
- [ ] tue_simulink_ethercat
- [ ] tue_teleop_keyboard
- [ ] upower_ros
- [ ] virtual_cam
- [ ] wire

LoyVanBeek, 2019-08-12:
For tue_robocup, I propose to split the PR into: 

- robot_smach_states+robot_skills, who'll get a thorough review
- separate PR for each challenge, which can be less thoroughly reviewed.

LoyVanBeek, 2019-08-12:
Checked above means PR merged or none needed?

MatthijsBurgh, 2019-08-12:
Both, Checked means no action needed anymore. I deleted all the unchanged rwc2019 branches already. And now we will start merging stuff, those will need to be checked, when a PR is merged.

closed 2020-01-04 by LarsJanssenTUe